signag/raspi-cam-srv #2: Problems when installing on Raspberry PI5 + setting different port than 5000

smurfxx, 2024-02-03:
During the installation, following your steps here https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#raspicamsrv-installation at step 9 the command "flask --app raspiCamSrv init-db" returns me this log:
`Traceback (most recent call last):
  File "/usr/bin/flask", line 33, in <module>
    sys.exit(load_entry_point('Flask==2.2.2', 'console_scripts', 'flask')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/flask/cli.py", line 1064, in main
    cli.main()
  File "/usr/lib/python3/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1651, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1698, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/flask/cli.py", line 579, in get_command
    app = info.load_app()
          ^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/flask/cli.py", line 309, in load_app
    app = locate_app(import_name, name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/flask/cli.py", line 236, in locate_app
    return find_best_app(module)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/flask/cli.py", line 64, in find_best_app
    app = app_factory()
          ^^^^^^^^^^^^^
  File "/home/pi/prg/raspi-cam-srv/raspiCamSrv/__init__.py", line 22, in create_app
    Path(app.instance_path + "/raspiCamSrv.log").touch(exist_ok=True)
  File "/usr/lib/python3.11/pathlib.py", line 1109, in touch
    fd = os.open(self, flags, mode)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/prg/raspi-cam-srv/instance/raspiCamSrv.log'
`

I also need to change the default listening port from 5000 to any other because the port 5000 is already used form Octoprint.
Thanks in advance!

signag, 2024-02-03:
Hi, thanks for your note.

The exception is a little bit strange since touch should create the log file if it does not exist.
Anyway, the file would only be required when using file handlers for logging which is not currently the case.

I have just pushed a version where the erroneous statement is commented.

I will also add a note related to the default Flask port.

Thanks

smurfxx, 2024-02-03:
Thank you for your fast reply! I'll wait for a release where I can set a custom port and I'll give you a feedback.

signag, 2024-02-03:
You have changed owner just for ~/prg/raspi-cam-srv/raspiCamSrv.
The database is, however, in ~/prg/raspi-cam-srv/instance  (this was also the location of the log file)
pi should be owner of ~/prg/raspi-cam-srv.

How did you create the folder ~/prg and below?
Because normally, when you create it as user pi, pi should have the ownership.

In the meantime, I have also adjusted README and the service template with respect to the Flask default port issue.
Hope it'll work then

smurfxx, 2024-02-03:
You're right, with the pi user I don't have the permission to create folders, so I create them with sudo. 
I changed the owner of raspi-cam-srv and customized the port: now everything works greats!

smurfxx, 2024-02-03:
Last request, forgive me!
I try to install the service as per your instructions, this is the content of the raspiCamSrv.service file:
[Unit]
Description=raspiCamSrv
After=network.target

[Service]
ExecStart=/home/pi/prg/raspi-cam-srv/.venv/bin/flask --app raspiCamSrv run --port 5001 --host=0.0.0.0
Environment="PATH=/home/pi/prg/raspi-cam-srv/.venv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin: /bin"
WorkingDirectory=/home/pi/prg/raspi-cam-srv
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

The service does not start and with the command
journalctl -u raspiCamSrv
I have this output

Feb 03 21:48:00 raspberrypi5 (flask)[121328]: raspiCamSrv.service: Failed to locate executable /home/pi/prg/raspi-cam-srv/.venv/bin/flask: No such file or directory
Feb 03 21:48:00 raspberrypi5 (flask)[121328]: raspiCamSrv.service: Failed at step EXEC spawning /home/pi/prg/raspi-cam-srv/.venv/bin/flask: No such file or directory
Feb 03 21:48:00 raspberrypi5 systemd[1]: Started raspiCamSrv.service - raspiCamSrv.
Feb 03 21:48:00 raspberrypi5 systemd[1]: raspiCamSrv.service: Main process exited, code=exited, status=203/EXEC
Feb 03 21:48:00 raspberrypi5 systemd[1]: raspiCamSrv.service: Failed with result 'exit-code'.
Feb 03 21:48:00 raspberrypi5 systemd[1]: raspiCamSrv.service: Scheduled restart job, restart counter is at 5.
Feb 03 21:48:00 raspberrypi5 systemd[1]: Stopped raspiCamSrv.service - raspiCamSrv.
Feb 03 21:48:00 raspberrypi5 systemd[1]: raspiCamSrv.service: Start request repeated too quickly.
Feb 03 21:48:00 raspberrypi5 systemd[1]: raspiCamSrv.service: Failed with result 'exit-code'.
Feb 03 21:48:00 raspberrypi5 systemd[1]: Failed to start raspiCamSrv.service - raspiCamSrv.

Where am I doing wrong?

signag, 2024-02-04:
Hi,
The error message sais that t it cannot find the Flasc executable under ```/home/pi/prg/raspi-cam-srv/.venv/bin```
This indicates that the virtual environment for ```/home/pi/prg/raspi-cam-srv/``` does not exist which would be located in the ```.venv``` subfolder.

My guess is that creation and activation of the virtual environment, as described in steps 7 and 8 went wrong.
Probably this was connected with the fact that your user pi does not have permission to create folders in his own home directory (which for me indicates that something must have gone wrong with your initial system setup).
When you have activated the virtual environment with ```source .venv/bin/activate```, your system prompt should have a ```(.venv)``` prefix.

1. First find the path of the flask executable with ```which flask```. This will probably be ```/usr/bin/flask```
2. Then check whether the folder ```/home/pi/prg/raspi-cam-srv/.venv``` exists.
3. If this folder exists, cd to ```/home/pi/prg/raspi-cam-srv``` and try to activate the virtual environment with ```source .venv/bin/activate```
4. If the prompt now has a ```(.venv)``` prefix, the virtual environment is activated and you can again search the path for the Flask executable with ```which flask```. The result should now reveal a path to a location in the virtual environment.

If your check shows that the virtual environment does not exist, you can just use flask from where you have installed it.
Just change the path in the service file to
```ExecStart=/usr/bin/flask --app raspiCamSrv run --port 5001 --host=0.0.0.0```
or any other path you found in step 1, above.
You should also change the Environment entry to
```Environment="PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin: /bin"```

Then you should be able to successfully start the service.

smurfxx, 2024-02-05:
Hi, after studying a bit about the setup of flask-based services I modified the service file like this:

[Unit]
Description=raspiCamSrv
After=network.target

[Service]
ExecStart=flask --app raspiCamSrv run --port 5001 --host=0.0.0.0
WorkingDirectory=/home/pi/prg/raspi-cam-srv
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

This setting works because the "flask" instruction, at least for my installation, works from any position so I simplified it by removing the Environment value and leaving only the WorkingDirectory and the execution command. So it seems to work, now I'll test your software a bit which is so far the only one working on Raspberry PI5!
(but I'll stress you a little more with a request for a particular function :) )

signag, 2024-02-05:
Hi,

obviously, you are not using the virtual environment and Flask 3 is installed in the standard environment.

The reason why I prefer and recommend using a virtual environment (see, fo example [Python/Virtual environment](https://wiki.archlinux.org/title/Python/Virtual_environment) or [venv](https://docs.python.org/3/library/venv.html)) is the following:
The Bookworm (or are you running on Bullseye?) distributions for Pi4 and Pi5 come with a preinstalled Flask 2.2.2.
If people would install Flask like you have done, this would be upgraded to Flask 3.0.0 and any applications which would require Flask 2 would no longer be working.

When installing Flask 3 in a virtual environment, as is recommended for raspiCamSrv, the existing Flask 2 installation is not touched.

Also, I plan to use other packages, like OpenCV together with raspiCamSrv and running in a virtual environment guarantees that there will not be any conflicts with other versions of these packages which might be either preinstalled or used by other software.

closed 2024-05-10 by signag