signag/raspi-cam-srv #3: New features request

smurfxx, 2024-02-05:
HI,
thanks to your software I can finally use the Raspi Cam V3 on Raspberry PI 5, I think it's the only application that can stream without problems, at least for the moment.
From my point of view you could implement small but interesting functions, such as:
- save the webcam configuration so if the service is restarted the settings are preserved
- disable the user registration tab once the first user registers
- put an option to make the stream accessible without access (i.e. the url /video_feed), it would be very useful for example for Octoprint which is set up to display the streaming from a specific url

I understand that your software is just beginning and you probably already have a lot of changes in mind but I hope my suggestions will inspire you! (if I knew how to program in Python I would gladly give you a hand but unfortunately it is not one of my known languages)

In the meantime, I'm already reporting your software in the Octoprint forum as I see it as the only solution for those who use that software ;)

sandyol55, 2024-02-05:
> The straming thread is automatically stopped and blocked while a raspiCamSrv user
takes a photo
takes a video
takes a photo series
In general, every use of the camera is kind of exclusive and I havn't yet found a possibility to do streaming parallel to other activities.

 If there is anything useful to be taken from my VERY simple attempt at a 'simultaneous stream and motion-triggered video record' please feel free !
 I hesitate to offer this very amateur first attempt at some code against your professional package but if it sparks any ideas then all is good!  
[https://github.com/sandyol55/Ropey-Cam](url)

signag, 2024-02-05:
> If there is anything useful to be taken from my VERY simple attempt at a 'simultaneous stream and motion-triggered video record' please feel free !

Thanks, I'll have a look

smurfxx, 2024-02-05:
> ## Streaming
> The URL for the video stream is: `http://<server>:<port>/video_feed` It is a MJPEG stream. When using this, you need to be, however, aware of the following:
> 
> * The feed is served by an own thread which will normally live as long as there is at least one client streaming
> * After 10 seconds with no one streaming, the thread stops
> * When there is a straming request and the thread is not active, it may take about 2 seconds until streaming starts
> * The straming thread is automatically stopped and blocked while a raspiCamSrv user
> * * takes a photo
> * * takes a video
> * * takes a photo series
> 
> In general, every use of the camera is kind of exclusive and I havn't yet found a possibility to do streaming parallel to other activities.

The conditions you listed are perfect for using a stream without login and this is what MJPEG stream still does, it would only be enough to guarantee anonymous access to the /video_feed endpoint and I would already be happy with that, given that the raspberry on where it is installed is never restarted and I can change the webcam settings on the fly with just a few clicks.
Thanks again for your support, sooner or later I will have to make a donation to you!

signag, 2024-02-05:
> it would only be enough to guarantee anonymous access to the /video_feed endpoint

Just comitted a version where the video_feed endpoint does not require login.

smurfxx, 2024-02-05:
> > it would only be enough to guarantee anonymous access to the /video_feed endpoint
> 
> Just comitted a version where the video_feed endpoint does not require login.

Thank you very much, I'll try it as soon as I can!

signag, 2024-02-18:
> * save the webcam configuration so if the service is restarted the settings are preserved

This function is now available.
See Chapter [Server Configuration](https://github.com/signag/raspi-cam-srv/blob/main/docs/Settings.md#server-configuration)

I have chosen not to automatically save every configuration change.    
Instead, you can prepare a configuration, explicitely save it and then spacify to use this at server startup.

signag, 2024-02-18:
> * disable the user registration tab once the first user registers

This function is now available.

See [raspiCamSrv Authorization](https://github.com/signag/raspi-cam-srv/blob/main/docs/Authentication.md)

lgeitner, 2024-02-19:
signag,

'save the webcam configuration so if the service is restarted the settings are preserved'

I have 2 cameras connected to my Raspberry PI 5, one camera is a camera module 3 (camera 0) and  the other camera is a camera module 3 NOIR (camera 1).

I have tested the store configuration option under settings, that you just added.  It doesn't seem to keep the settings upon reboot

Here is how I configured the settings:

under the Settings tab, 

active camera  0:
I changed video type to h264
I clicked on Submit.

I then clicked on Config Tab
under Live View Tab
I changed Sensor Mode: to Sensor Mode 2
I clicked on Submit & Apply

I clicked on the Live tab --> Focus
I checked Autofocus Mode and changed to Continuous
I clicked on Submit

I went back to Settings tab 

I changed to active camera 1:
I changed video type to h264
I clicked on Submit.

I then clicked on Config Tab
under Live View Tab
I changed Sensor Mode: to Sensor Mode 2
I clicked on Submit & Apply

I clicked on the Live tab --> Focus
I checked Autofocus Mode and changed to Continuous
I clicked on Submit

I went back to the Settings Tab
I checked the option to Start server with stored Configuration:
I clicked on Store Configuration and then clicked on OK.

I rebooted my Raspberry PI 5

I connected back into my Raspberry PI 5 and ran the following commands

cd ~/prg/raspi-cam-srv
source .venv/bin/activate
flask --app raspiCamSrv run --port 5000 –-host=0.0.0.0

All the settings are back to default expect for the settings under Live --> Focus --> Autofocus Mode: Continuous.

Am i doing something incorrectly?

Thank you for implementing the save configuration option!

Lance

smurfxx, 2024-02-19:
> This function is now available. See Chapter [Server Configuration](https://github.com/signag/raspi-cam-srv/blob/main/docs/Settings.md#server-configuration)
>
> I have chosen not to automatically save [...]

I made a pull of your git project, now I see your new setting:
- "Storing configuration" and "Load stored configuration" works
- "Start server with stored Configuration" does not work, neither if I start the server manually nor as a service, I see that the _loadConfigOnStart.txt marker file is not created, I tried to create it manually but even in this situation the server does not load the stored configuration on load. 

The "Start server with stored Configuration" option even if it doesn't work is not a big problem, I just load the stored configuration and that's it ;)

signag, 2024-02-19:
I have fixed this.

I had usually tested with some photos in the display buffer.
When the display buffer was empty, this caused an error when loading the serverConfig.json file.

In addition, there was another issue: While collecting camera-specific information, which is always done because camera setup may have changed, the sensor mode and stream size have been overwritten with default values for the active camera.
This has also been fixed.

I also tested it now with 2 cameras where one was a camera module 2 without focus.

Thanks for your support

smurfxx, 2024-02-19:
> I have fixed this.
>
> I had usually tested with some photos in the display buffer. When the display buffer was empty, this caused an error when loading the serverConfig.json file.
>
> In addition, there [...]

I confirm that now it's fixed, thank you for **YOUR** support!

lgeitner, 2024-02-19:
I did a git pull of your project.

The Sensor Mode setting seems to defaults back to Custom all the time.

If I switch from one camera to another under settings and then click Submit
If I then go to the Config -> Live View setting.
The Sensor Mode setting seems to defaults back to Custom.
It does not seem to retain the setting of Sensor Mode 2 that I have previously selected and then clicked on Submit and Apply.

Thank you for work on this project!

signag, 2024-02-19:
> If I switch from one camera to another under settings and then click Submit
> If I then go to the Config -> Live View setting.
> The Sensor Mode setting seems to defaults back to Custom.

I have implemented it like this on purpose.
The reason is that different cameras have different sensor modes and usually sensor modes with the same number have completely different properties on different cameras.
An exception are the module 3 cameras in normal, wide and NoIR variants which all have the same sensor.

However module 1 cameras have 4 sensor modes, module 2 cameras have 7 and module 3 cameras have 3.
I currently do not know about HQ and Global Shutter.
And also the properties like pixel size, format or crop limits are completely different for sensor modes with the same number for camera modules with different sensors.

Therefore, it would normally not be a good idea to keep the sensor mode when changing the camera, exept when both cameras have the same sensor.

Tho logic for initially selecting sensor mode and stream size for the different use cases is currently as follows:
- For Photo and Raw Photo take the highest sensor mode, which is normally the one with the highest resolution
- For video take the first sensor mode [0], which is the one with the lowest resolution and highest frame rate
- For the live stream set the resolution to a width of 640px and determine the height according to the sensors aspect ratio.<br>Then try to find a sensor mode with the same width and height (for example for module 1 camera this would be sensor mode 0 with 640x480px)<br>If no sensor mode fits the size, set sensor mode to "Custom"<br>With "Custom" you can freely set the stream size independently from the sensor modes (This is not possible for Raw Photo)

@lgeitner, that is the reason why you always see sensor mode "Custom" when you switch between your two module 3 cameras.

lgeitner, 2024-02-19:
ok.  Thank you for the explanation.  I will try what you suggested above.

Thanks,
Lance

signag, 2024-02-21:
> If there is anything useful to be taken from my VERY simple attempt at a 'simultaneous stream and motion-triggered video record' please feel free !

Hi @sandyol55,
thanks for pointing me to your solution which is a nice little application with some interesting aspects.

Looking at the question of having a live stream simultaneously to video or image capture, this works in your case as you are using one single camera configuration for all of your threads.

As soon as the configuration for live stream and photo or video capture are different, this is no longer possible as far as I understand picamera2.

My initial idea was to allow for independent configurations for the different use cases, with live stream being one of them.

But I know that it is possible in picamera2 to run different encoders in parallel on the main and lores streams, which would allow using the lowres stream for MJPEG streaming while recording video with the main stream.   
However, this would require using one single configuration.

For a motion capture solution, it makes absolutely sense to have a continuous live stream.

So, your solution gives me some inspiration.

But let me ask a question:   
With the START button you start recording with the MJPEGEncoder.   
From my understanding, the MJPEGEncoder will now encode the main stream, because encode='main' is the default for create_video_configuration.   
However, the StreamingOutput, resulting from this decoder, is never used.   
Instead, you encode the lores straem with simplejpeg in the mjpeg_thread.

Would it not have been possible to use the StreamingOutput for the live stream.
You would then have to set encode='lores' in create_video_configutation.   
And in motion(), you would have to start the encoder with ```picam2.start_encoder(encoder, name='main')```

The 'name' parameter is mentioned in the picamera2 manual but never explicitely showing up in an example.   
I also never used it myself.

So, thanks again for this hint.   
Maybe, I will add motion-triggered video capture to raspiCamSrv, as you had asked earlier.

sandyol55, 2024-02-21:
Many thanks for taking the time to have a look through the 'spaghetti ! '  code. ;-)

The code added to use the buttons was very much an add-on once I'd figured out how to use post !
Not sure why I chose the MPEGEncoder to stop/start, when the video recording is intended to be with the H264Encoder and FFmpegOutput, I'm sure I had a reason at the time but can't recall it now!

I will go through some of the inconsistencies you have found and will try to tidy things up and incorporate your comments/ideas - just for my own education if nothing else!!

If motion-triggered capture was ever added I think there would be a lot of RPiCamWeb , PiKrellCam and Motioneye users who would be keen to try out raspiCamsrv, to be able to use newer cameras and a libcamera/picamera2 based package in place of those now unsupported legacy stack based packages.

Again thanks and best wishes for your progress with raspiCamSrv.

signag, 2024-03-16:
> If motion-triggered capture was ever added

This is now available (since V2.2.0), at least in a basic version.

closed 2024-05-10 by signag