signag/raspi-cam-srv #5: Using raspiCamSrv with OctoPrint

jesternberg, 2024-02-24:
I am using raspiCamSrv with RPI5 and 2 cameras.  Is it correct to have two versions of the service running on different ports to be able to use the multicam plugin? or is there a better solution.

Also since snapshots are not currently exported you cannot configure the snapshot url in OP.  I tried the following path, based on video_feed, but it is not quite right.  Using mimetype='multipart/x-mixed-replace, I do see the snapshot directly in a browser.  OP is expecting the mimetype to be image/jpeg but with that as the mimetype, both OP and browser do not error but image is not displayed.  Any suggestions as to what is wrong?

diff --git a/raspiCamSrv/home.py b/raspiCamSrv/home.py
index c0ccb85..f6cc62f 100644
--- a/raspiCamSrv/home.py
+++ b/raspiCamSrv/home.py
@@ -48,6 +48,23 @@ def video_feed():
     return Response(gen(Camera()),
                     mimetype='multipart/x-mixed-replace; boundary=frame')
 
+def gen2(camera):
+    """Snapshot generator function."""
+    #logger.debug("In gen2")
+    yield b'--frame\r\n'
+    frame = camera.get_frame()
+    #l = len(frame)
+    #logger.debug("Got frame of length %s", l)
+    yield b'Content-Type: image/jpeg\r\n\r\n' + frame + b'\r\n--frame\r\n'
+
+@bp.route("/snapshot")
+# @login_required
+def snapshot():
+    #logger.debug("In snapshot")
+    return Response(gen2(Camera()),
+                    mimetype='image/jpeg; boundary=frame')
+                    #mimetype='multipart/x-mixed-replace; boundary=frame')
+
 @bp.route("/photos/<photo>")
 @login_required
 def displayImage(photo: str):

signag, 2024-02-26:
> Is it correct to have two versions of the service running on different ports to be able to use the multicam plugin? or is there a better solution.

Probably, for now you would need duplicate installations of raspi-cam-srv.
A better solution would be, of course, to allow for an additional streaming endpoint for the second camera.

I could think of a page in raspiCamSrv showing live streams of both cameras simultaneously.
Then, you could use both endpoints for streaming.

I am currently refactoring in order to support an uninterrupted live stream while taking pictures or videos.   
This would also be a precondition for a snapshot endpoint.
This will take a little bit. Therefore, I do not currently know when I will get to new features.

signag, 2024-02-26:
> since snapshots are not currently exported

The best solution would be to provide endpoints for snapshots for both cameras.

I'll put that on the list.

signag, 2024-03-03:
The endpoint for photo snapshots for the active camera is now available in V2.1
See [Web Cam Settings](https://github.com/signag/raspi-cam-srv/blob/main/docs/Webcam.md)

Endpoints for the second camera will follow later

signag, 2024-03-16:
> Is it correct to have two versions of the service running on different ports to be able to use the multicam plugin? or is there a better solution.

With version V2.3.0 one server instance supports streaming for both cameras sumultaneously.

bradley1999, 2024-03-20:
Problem viewing live stream if any changes are made on another screen. Live feed stops and can't be viewed unless program is stopped and restarted. On Terminal only message is "WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise."
Any ideas why the live feed stops and can't be restarted?
Also the service configuration change doesn't work after installation. Change shows user name, but closing an SSH session stops the feed. 
Using Rpi 4 with latest 64 Bookworm, official Camera module 3. Viewing on Macbook Pro using Firefox. Same problem on Safari and Chrome.

signag, 2024-03-20:
> Problem viewing live stream if any changes are made on another screen.

Which changes did you make which caused the live stream to disappear?    

Sometimes, a specific camera configuration (Sensor Mode, Stream Size or others) may lead to errors within Picamera2, for example in the decoder. Typically, there should be an exception thrown by the relevant program.    

Did you see any error messages in addition to the warning which you have mentioned?

If you run raspiCamSrv as service, you would need to look into the journal.

I am working on a solution to make raspiCamSrv more robust against errors in these backgound processes.

signag, 2024-03-20:
> Also the service configuration change doesn't work after installation.

Could you please specify what exactly you mean with 'service configuration change'?

Which of the [Service Configuration](https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#service-configuration) procedures did you follow (with or without Audio Support)?

bradley1999, 2024-03-20:
On the initial bootup, ipaddress:5000, live stream is shown, but only Photo, Raw, Video black control buttons. No Hide or Clr.
At top right red stream icon is on.
Take photo, nothing happens. Click on Raw or Video, same negative response.
No errors thrown on Terminal. 
When browser refreshed, still no changes.

Stop running with Ctrl-C, then restart causes live stream to appear, with Hide and Clr controls, plus photo below and metadata. Red stream icon is on.

Move to Trigger settings to start motion detection, red movement icon is on, but any movement in front of camera is ignored. Can't take photo or video.

I'm using a 16Gb SSD card, is this too low and would a larger memory make a difference?

When I close a SSH session, the screen is lost with a can't connect message, so Raspicamsvr has closed down. I wanted to keep it running in the background without having to restart each time. 

Added a few screenshots 
[image]

[image]

[image]

bradley1999, 2024-03-20:
Live view config screenshot.
[image]

bradley1999, 2024-03-20:
live view config screenshot
[image]

signag, 2024-03-20:
> I wanted to keep it running in the background without having to restart each time.

To keep the Flask server running independently from the SSH session, you need to configure it as service.    
The description is under [Service Configuration](https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#service-configuration).    
Did you have issues with this?

Thanks for the other info.   
I could partly reproduce the issue with video taking and with motion detection.   
Taking photo and raw photo worked.    
However, I have a camera model 1 currently connected to the Pi4. Trying now with a model 3.

bradley1999, 2024-03-20:
For independent running without audio, I followed the info to install as service. 
A check of this line
sudo journalctl -e
gave this response, appears to start then stop several times, before starting. I have rebooted.
When the SSH session closes, so does Raspicamsrv and laptop browser shows no connection.

[image]

signag, 2024-03-20:
Please post the entire error description ```(flask)[3357]: ...```

bradley1999, 2024-03-20:
Do you need just this one line expanded, or the full 1000 lines!

signag, 2024-03-20:
Just two lines (flask)[xxxx] with the same number xxxx

bradley1999, 2024-03-20:
I wonder if there is a permissions issue. Configuring service without audio shows these lines

nano ~/raspiCamSrv.service
and
sudo cp ~/raspiCamSrv.service /etc/systemd/system

In the first line it shows open a SSH session, but is this before moving to the repository and virtual environment?
I also deleted <user> and added user name jules, but without the <> brackets. Are the brackets required? I'm sure they're not.....

I

signag, 2024-03-20:
If you don't mind, would you show the content of the service file:     
```cat /etc/systemd/system/raspiCamSrv.service```

Here is mine (for user 'sn' which should be 'jules' in your case:
[image]

bradley1999, 2024-03-20:
Comparing mine to yours I can see <user> twice, but no idea why as I added jules in the nano configuration

signag, 2024-03-20:
There are two more places where you need to replace ```<user>``` with ```jules```

bradley1999, 2024-03-20:
I also couldn't expand the lines you asked for. Can't move right to see the text of the full line.
Have repeated the process, but different higher line numbers appear, I'm guessing as it's a different session

bradley1999, 2024-03-20:
Yes I can see the two errors now, but how to change them? Where are the two other locations found? Using a nano session as in your description, only one location is shown as needing changing.

signag, 2024-03-20:
Normally, when you have the file open in nano (remember to sudo when editing in ```/etc/systemd/system```), you just  
  
- place the cursor in front of the text you want to replace
- delete the old text (Del key)
- and enter the new text

bradley1999, 2024-03-20:
Thanks for the help. Changed user name in the two extra places and I think that's done it!
Closed SSH session and checked web browser which shows this

[image]

bradley1999, 2024-03-20:
Now 
sudo journalctl -e
gives this (last section as there are a lot of lines)

[image]

bradley1999, 2024-03-20:
Is that what is expected....how does it compare with your response to 
sudo journalctl -e ?

signag, 2024-03-20:
You should see this:    
[image]

In your screenshot, I do not see the service start.

Probably, you need to restart it:   
```sudo systemctl stop raspiCamSrv.service```    
```sudo systemctl start raspiCamSrv.service```

bradley1999, 2024-03-20:
Here's what I can see on Terminal startup 
[image]

bradley1999, 2024-03-20:
here's my startup on Terminal
[image]

signag, 2024-03-20:
The bottom line just refers to your user session and not to the Flask server.   
You need to look at the red error messages with [Flask]

They say that here is no file ```/home/jules/prg/raspi-cam-srv/.venv/bin/flask```   
This refers to the Flask executable.
You may chack yourself:   
```cd ~/prg/raspi-cam-srv/.venv/bin```   
```ls -al```   

If the output looks like     
[image]    
your setup is OK.

If you do not see ```flask```, you did not install Flask within the virtual environment as has been requested in step 9 of the [RaspiCamSrv Installation](https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#raspicamsrv-installation).   

You may check where Flask is installed:   
Make sure that the virtual environment is deactivated. Then enter   
```which flask```   
You will see   
[image]

Now, activate the virtual environment and repeat the which command:   
[image]

If your output is still ```/usr/bin/flask```, then Flask is not installed where it should be.   

Then you can repeat step 9 of the [RaspiCamSrv Installation](https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#raspicamsrv-installation) and everything should be fine.

signag, 2024-03-20:
> here's my startup on Terminal <img alt="Screenshot 2024-03-20 at 19 35 51" width="849" src="https://private-user-images.githubusercontent.com/16852298/314648215-368e7ebe-0ed9-4928-9891-9abdfbb10cf1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NjM3MTYsIm5iZiI6MTcxMDk2MzQxNiwicGF0aCI6Ii8xNjg1MjI5OC8zMTQ2NDgyMTUtMzY4ZTdlYmUtMGVkOS00OTI4LTk4OTEtOWFiZGZiYjEwY2YxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE5MzY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3NWU2MTg3NzYxMmQ2ODQ4N2I4YTE5M2NiNzQ4YzhiNGZmYzQyNGJhOTAxNjc5Zjk3ZWQwN2ZiMTFmNmUxMzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Vt4QCUysLPjA5zZzlQplr2-65nn_0_E2WZh7TZ7H_Vw">

This does not tell you that you have started Flask from your virtual environment.   
The system will find Flask also at ```/usr/bin``` because this is in the Path.    

The service script, however, expects that Flask is installed in your virtual environment.

bradley1999, 2024-03-20:
after ls -al get this
[image]

bradley1999, 2024-03-20:
seems to be missing 4 lines compared to yours. Followed the installation step by step.

fonttools
pyftmerge
pyftsubset
ttx

are these missing lines responsible for my live view missing probs?

signag, 2024-03-20:
No

signag, 2024-03-20:
Please show again the content of the service file:
```cat /etc/systemd/system/raspiCamSrv.service```

The error message before also complained about the working directory.

bradley1999, 2024-03-20:
Which flask responds correctly.

I'm still seeing <user> twice, despite changing it and rebooting!
[image]

bradley1999, 2024-03-20:
still seeing USER twice

signag, 2024-03-20:
Probably, you did not save the file or did not open with sudo:   

```sudo nano /etc/systemd/system/raspiCamSrv.service```   

Then make the changes

Then save with "^O" (<Ctrl + O>)   
Hit <Enter> to confirm the file name:    
[image]

Then exit with "^X"

Then check again with    
```cat /etc/systemd/system/raspiCamSrv.service```

signag, 2024-03-20:
In the meantime, I have committed some fixes which should cover most of the issues described earlier for Pi4 with Cam3.   
The reason was that the encoder failed with higher resolutions.   
Now, the resolutions for video are set to lower values for Pi4, similar as for Pi Zero.

To update:    
```cd ~/prg/raspi-cam-srv```    
```git pull```

bradley1999, 2024-03-20:
Ok so redid the changes which had these results
[image]
[image]

bradley1999, 2024-03-20:
[image]

bradley1999, 2024-03-20:
[image]

bradley1999, 2024-03-20:
after changes now I see
[image]

bradley1999, 2024-03-20:
How do I change port 5000 which is now in use.....not sure why?

signag, 2024-03-20:
It seems that the raspiCamSrv service is now active. This will of course use port 5000.

Did you check with   
sudo journalctl -e  ?

Did you try to connect?

bradley1999, 2024-03-20:
yes, it connects to port 5000 but i'd like it to be another port.
Motion was running, but trying to stop it caused an error. The Stop button had no effect.
1000 lines of error code....here is the last part
[image]

bradley1999, 2024-03-20:
The updated fixes seem to have changed something...the Hide and Clr buttons below the live window have returned, but onpressing Photo, no pic is taken.

signag, 2024-03-20:
The Hide and Ctrl buttons are only visible if a photo has been taken and shown in the lower area. See the user guide.

So, if you see it, you must have taken a photo.

signag, 2024-03-20:
If you want to use another port, you just change 5000 to the port you want to use in the service script or when you start the Flask server directly

bradley1999, 2024-03-21:
Tried to change port 5000 which was apparently in use, not sure what else is using it.
[image]

bradley1999, 2024-03-21:
So followed instructions, changing it to port 5001 in Flask and url address.
[image]

 Which gave this on webpage
[image]

and a long error readout
[image]
[image]

signag, 2024-03-21:
If you look at the error message, you see "Device or resource busy"    
This means that you have another process which uses the camera.

You cannot have multiple processes using the same camera.

I do not know what you have running on your system.   
It seems that the raspiCamSrv service is running in the background

You can either run the service or start the Flask server manually.   
Not both at the same time.

If you want to start manually, you need to stop the service first:    
```sudo systemctl stop raspiCamSrv.service```    

Remember: when booting the device, the service will start automatically.

bradley1999, 2024-03-22:
That's all fine, seems to be up and running, but now have a password problem! On first login after powering off, can't login with current password, and doesn't seem a way to reset it. The option to change password asks for current password.....

signag, 2024-03-22:
The simplest way is to reset the database, which will remove the user and force new registration.

However, currently this will also reset the database for motion tracking.
If you wanted to avoid this, the only way would be to use a SQLite client to access the database and remove the user entry in table  'user'.

If you don't mind losing all motion tracking events, proceed as follows:
```sudo systemctl stop raspiCamSrv.service```
```cd ~/prg/raspi-cam-srv```
```source .venv/bin/activate```
```flask --app raspiCamSrv init-db```
```sudo systemctl start raspiCamSrv.service```

bradley1999, 2024-03-22:
Thanks, that's fixed it, don't need a record of motion. I realise a problem of my own making which produced the 5000 "device or resource busy" error........I had RaspiCamSrv running on boot using 5000, and then used service to restart it, so two processes running on same ip address.

I have a lot of test photos during testing which aren't needed, how do I delete them all?

closed 2024-08-27 by signag