signag/raspi-cam-srv #6: Raspberry Pi 5 x64 Bookworm with Camera Module 3 - No cameras detected

jonathan88n, 2024-03-05:
When starting up the camsrv, I see error logs in the terminal that says the following:

WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:5000
 * Running on http://192.xxx.xx.xx:5000
Press CTRL+C to quit
[2024-03-04 21:52:16,600] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/workspace/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 867, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 852, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/raspi-cam-srv/raspiCamSrv/auth.py", line 273, in wrapped_view
    return view(**kwargs)
           ^^^^^^^^^^^^^^
  File "/workspace/raspi-cam-srv/raspiCamSrv/home.py", line 25, in index
    Camera().startLiveStream()
    ^^^^^^^^
  File "/workspace/raspi-cam-srv/raspiCamSrv/camera_pi.py", line 878, in __new__
    activeCam = Camera.getActiveCamera()
                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/raspi-cam-srv/raspiCamSrv/camera_pi.py", line 1054, in getActiveCamera
    raise SystemError("No cameras were found on the server's device")
SystemError: No cameras were found on the server's device
[2024-03-04 21:55:12,597] ERROR in camera_pi: Thread 140735407976832: Camera._thread - Exception.

[image]
[image]
I've attached two pictures of the cable hook up into cam0 slot.

signag, 2024-03-05:
Your cabeling looks OK.

The error is raised when Picamera2 does not return any camera information.
You can test it directly:
```
~ $ python
>>> from picamera2 import Picamera2
>>> print(Picamera2.global_camera_info())
>>> quit()
```

After some libcamera INFOs, you should see      
```[{'Model': 'imx708_wide_noir', 'Location': 2, 'Rotation': 0, 'Id': '/base/soc/i2c0mux/i2c@1/imx708@1a'}]```

If you see an empty list, instead   
```[]```,   
Picamera2 did not find a camera.

In this case, you may have activated legacy camera support.
Open the Raspberry Pi Configuration tool:

```~ $ sudo raspi-config```   
Select "3 Interface Options"   
Then "I1 Legacy Camera Enable/disable legacy camera support"

If you see "Would you like to enable legacy camera support?"   
select ```<No>```

When back in the main menu, choose ```<Finish>```

The tool will request to reboot, which must be confirmed with ```<Yes>```

After reboot, you can reconnect via SSH and repeat the direct test with Python, described above.

jonathan88n, 2024-03-05:
print(Picamera2.global_camera_info()) did return an empty list ( [] ).

I did sudo raspi-config and went to Interface Options but had no Legacy or Camera options:
[image]

signag, 2024-03-05:
Can you please check ```/boot/firmware/config.txt```

This must have an active entry ```camera_auto_detect=1```

jonathan88n, 2024-03-05:
sudo cat /boot/firmware/config.txt | grep camera_auto

camera_auto_detect=1

ZombieToad, 2024-03-08:
> ![rpi5-1](https://private-user-images.githubusercontent.com/957397/309968114-20264a51-7532-4fdb-9fd3-b1ff598506aa.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk5MzQyMzksIm5iZiI6MTcwOTkzMzkzOSwicGF0aCI6Ii85NTczOTcvMzA5OTY4MTE0LTIwMjY0YTUxLTc1MzItNGZkYi05ZmQzLWIxZmY1OTg1MDZhYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwOFQyMTM4NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMDc2OGJhMWJjZDIzM2JkNDA1MGE5NDEwMmQzM2IwNzNmOGUyZjMzN2I3ZTg2MDZhZjE3ZDE4MDc3NGIyNjFjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.6bukpTSfkt7rAkoJFYLsCXBOjDGyWdth9mTJ0SePliE)

The cable connected to your Raspberry Pi 5 is reversed

jonathan88n, 2024-03-11:
@signag, I have indeed looked elsewhere but I'll give those troubleshooting steps a try.
@ZombieToad, are you saying the connector (gold side) should be facing the HDMI instead of facing where the SD card is inserted? I read elsewhere that that cable should be facing opposite direction of the HDMI port.

ZombieToad, 2024-03-11:
> @signag, I have indeed looked elsewhere but I'll give those troubleshooting steps a try. @ZombieToad, are you saying the connector (gold side) should be facing the HDMI instead of facing where the [...]

the HDMI and SD card slot are on the same side from the camera connector? the gold contacts should be facing the ethernet port (away from the hdmi ports)
[image]

jonathan88n, 2024-03-11:
@ZombieToad, so I tried flipping the cable around to face the Ethernet port but still no luck:
[image]
[image]
[image]

@signag, I noticed in those troubleshooting steps, the permission check shows I don't even have /dev/video0 at all:
[image]

[image]

signag, 2024-03-11:
What about a fresh install with latest imager version 1.8.5 from [https://www.raspberrypi.com/software/](https://www.raspberrypi.com/software/)?

What is your OS version?

Here is mine:
[image]

jonathan88n, 2024-03-11:
Here is my OS info:
[image]

I just recently did a fresh image 2 weeks ago.

signag, 2024-03-11:
I still think that your cables might be pushed in further.

here is my setup:

[image]

[image]

Do you have the possibility to try with another camera?   
Perhaps this one has a defect.

jonathan88n, 2024-03-11:
So, I do have another Camera Module 3 camera that works successfully with a RPI4 x64 Bulleye and I have tried to use that camera on the RPI5 and the cable doesn't not go any further (it hits a wall, if you will) when inserting it.
Both cameras do not get detected in the RPI5 while both DO get detected in the RPI4.

signag, 2024-03-11:
Then I have no more idea, except that the camera port on your pi5 does not work.

jonathan88n, 2024-03-11:
Does /dev/video0 and /dev/video1 show for you whether or not you have camera plugged in for you?
ls -lha /dev/video* 
doesn't return any video0 or video1, only video19+

signag, 2024-03-12:
This is with no camera plugged in:     
[image]

This is with one camera plugged in:     
[image]

And this with two cameras:    
[image]

jonathan88n, 2024-03-12:
What cable connector are you using between your RPI and Camera Module 3 (link if you don't mind)? I will try getting a new cable

signag, 2024-03-13:
I use this one:    
[https://www.rasppishop.de/Raspberry-Pi-Zero-Kamera-Kabel-150mm-11944](https://www.rasppishop.de/Raspberry-Pi-Zero-Kamera-Kabel-150mm-11944)     

and this:
[https://www.rasppishop.de/Raspberry-Pi-5-Kamera-Kabel-Standard-Mini-500mm](https://www.rasppishop.de/Raspberry-Pi-5-Kamera-Kabel-Standard-Mini-500mm)

spookytheripper, 2025-02-14:
I tried just today to fresh install it on a new bookworm install. 

The Camera v3 **wide** gave me a lot of similar issues, it is somehow detected (it shows up in the config as "imx_708_wide") but it can't be initialized. I know the camera works because i could take a test pic with a simple "libcamera-jpeg -o test.jpg"

When i load the live preview page it starts by giving me an error 500, then if i refresh i get the following error:

Error in Camera._thread: Error in live view: CameraController.requestCameraForConfig - Camera could not be started
Probably a different camera configuration can solve the problem.

Though i'm not sure where i can try other camera configurations...

I then Switched to a Camera v3 "normal/non-wide" and it worked with out issues. if i try to switch back to wide, it stops working again.

PS: i also needed to install ffmpeg separately to record videos

nigeljohnson73, 2025-06-15:
If it helps anyone, I had a problem with libcamera0 that I was installing in my initial set up. When I then installed the rpicam-apps I could only see cameras that were plugged into port 1 (the furthest from the ethernet port).

Many fresh installs later, I found that if I tried to install libcamera0 and rpicam-apps together it told me there was a problem with libcamera. I removed the libcamera0 from the install line and rpicam-apps installs its dependencies correctly, including a working libcamera. All the other stuff I set up also worked fine.

Now my fresh install works properly, and I can use port 0 as well as port 1.

damienwolf07, 2025-09-16:
This issue seems to be resolved, This issue can be closed.