signag/raspi-cam-srv #9: Camera not showing in Live

wyohman, 2024-04-07:
Brand new install on bookworm. I tested camera with libcamera-hello:

wyohman@raspberrypi:~ $ libcamera-hello
[0:04:34.782090969] [873]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:04:34.954627653] [876]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:04:34.960985705] [876]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media1
[0:04:34.961190133] [876]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:04:34.966523388] [873]  INFO Camera camera.cpp:1183 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-SBGGR10_CSI2P
[0:04:34.967358705] [876]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
#0 (0.00 fps) exp 30141.00 ag 2.00 dg 1.06
#1 (30.01 fps) exp 29981.00 ag 1.97 dg 1.10

As well as libcamera-still -o test.jpg and the jpg looked very nice.

Updated config to start automatically and I was able to create a user and login.  When Iogin I can see the right side options Focus, Zoom, etc.. but nothing on the Live side.

If I click Photo, Raw or Video, I see this message at the bottom (message shown for each button).

Error in Camera.takeImage: Phototaking caused error: [Errno 12] Cannot allocate memory
Error in Camera.takeRawImage: Taking raw photo caused error: [Errno 12] Cannot allocate memory
Error in Camera.takeImage: Phototaking caused error: [Errno 12] Cannot allocate memory

Feels like I must missing something but I went over the instructions multiple times.

Info shows what I would expect for my setup (RPI 3b+ and imx708_wide)

signag, 2024-04-07:
It seems you are using a Pi 3.
Unfortunately, I havn't one myself for testing.
My experience with Pi Zero and Pi 4 was that these systems had memory issues with the larger resolution of the new sensors and larger values for buffer_count. Also, these systems require "YUV420" format for the lowres stream.     
Therefore, I set the default values for configuration settings accordingly.

Checking in my code, I just see that the buffer_count for Pi 3 is not set to a lower value of 2 as it is done for Pi Zero and Pi 4.

Can you check in the *Config* page for *Live View* which value for *Buffer Count* is set.    
If the value is 6, set it to 2.      
The settings for Pi 3 should probably be the same as for Pi 4:     
[image]

signag, 2024-04-07:
... and for *Video*, *Buffer Count* should be 4.

signag, 2024-04-07:
I just staged version V2.3.6 where the settings should also apply for Pi 3.

You can update with    
```cd ~/prg/raspi-cam-srv```
```git pull```     
```sudo systemctl restart raspiCamSrv.service```

wyohman, 2024-04-07:
WOW! Thanks for the info and update! I now see live view and will be working on putting it in place for more testing.

Where can I donate to keep this going?

closed 2024-05-10 by signag