signag/raspi-cam-srv #13: Error Connecting To Server: sqlite3.OperationalError: no such table: user

JS1Dev, 2024-05-07:
Following the [RaspiCamSrv Installation](https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#raspicamsrv-installation), I got to the following step (#14)
```
Connect to the server from a browser:
http://<raspi_host>:5000
This will open the [Login](https://github.com/signag/raspi-cam-srv/blob/main/docs/Authentication.md#log-in) screen.
```
When I went to my browser to specify: http://<raspi_host>:5000, I got the following:

```
(.venv) <redacted>@<redacted>:~/prg/raspi-cam-srv $ flask --app raspiCamSrv run --port 5000 --host=0.0.0.0
 * Serving Flask app 'raspiCamSrv'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:5000
 * Running on http://<redacted>:5000
Press CTRL+C to quit
[2024-05-07 17:46:02,945] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/home/jim/prg/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jim/prg/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jim/prg/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 865, in full_dispatch_request
    rv = self.preprocess_request()
         ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jim/prg/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 1239, in preprocess_request
    rv = self.ensure_sync(before_func)()
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jim/prg/raspi-cam-srv/raspiCamSrv/auth.py", line 235, in load_logged_in_user
    g.nrUsers = get_db().execute("SELECT count(*) FROM user").fetchone()[0]
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: no such table: user
<redacted IP> - - [07/May/2024 17:46:02] "GET / HTTP/1.1" 500 -
[2024-05-07 17:46:03,068] ERROR in app: Exception on /favicon.ico [GET]
Traceback (most recent call last):
  File "/home/jim/prg/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jim/prg/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jim/prg/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 865, in full_dispatch_request
    rv = self.preprocess_request()
         ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jim/prg/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 1239, in preprocess_request
    rv = self.ensure_sync(before_func)()
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jim/prg/raspi-cam-srv/raspiCamSrv/auth.py", line 235, in load_logged_in_user
    g.nrUsers = get_db().execute("SELECT count(*) FROM user").fetchone()[0]
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: no such table: user
<redacted IP>- - [07/May/2024 17:46:03] "GET /favicon.ico HTTP/1.1" 500 -

```
Possibly a missing step in the [RaspiCamSrv Installation](https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#raspicamsrv-installation) to a initialize the creation of the table?  If so, what might that step be and should it be run from the virtual environment activated (I'm a bit new to virtual environments)?

signag, 2024-05-08:
The step to initialize the database is 11:

Initialize the database for Flask
(with raspi-cam-srv as active directory and the virtual environment activated - see step 7):
```flask --app raspiCamSrv init-db```

JS1Dev, 2024-05-09:
That worked, thanks :).  

Well... that was embarrassing, not sure how I followed all those instructions and just missed that one.  I even ran into and resolved the ModuleNotFoundError, with the help of your instructions, and did the Optional step 10.  fwiw, your instructions are quite good, again, not sure how I missed that one step.  In fact, your instructions introduced me to creating a virtual environment, which I've gone on to read about the benefits of and actually used for something else today.  I case you couldn't tell, I'm new to Linux - I've been teaching myself on the weekends for about a month with a Windows/Ubuntu dual boot.

For context, my interest/hope in your software is just to allow me to see what my Module 3 Pi Cam (my first one) is capable of (new to Pi too), and in the short time I've played with it it looks like it does exactly what I hoped.  Honestly, it seems like a perfect companion for anyone buying a Pi Cam.

One thing it showed me was that I can pan in directions and zoom, which is what I'd like to do in Frigate (also new to me).  I'm also currently reading through the [picamera2-manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) to see if I can configure my Module 3 Cam to maximize its capabilities in Frigate via [MediaMTX](https://pkg.go.dev/github.com/aler9/rtsp-simple-server).

One curiosity question, I was just wondering if you've ever thought about enabling RaspiCamSrv in a Docker container?  btw, I have no idea how challenging that might be, so grain of salt with that suggestion.  I ask because I've become pretty addicted to Docker since using Linux, I absolutely love it and docker compose, I even installed it on my Pi so I could install MediaMTX in a container - actually virtual environments kind of reminds me of it a little.

Anyway, thanks for your quick response, introducing me to virtual environments, and your cool software :)

closed 2024-05-10 by signag