signag/raspi-cam-srv #14: Selecting "1: imx296" as active camera and then clicking submit kills the server

Hermann-SW, 2024-05-08:
Details here:
https://forums.raspberrypi.com/viewtopic.php?p=2219364#p2219364

Besides that raspi-cam-srv is really nice with my two GS cameras on Pi5:
[image]

signag, 2024-05-08:
I could reproduce the error. 
Need to analyze.

It seems that Picamera2 wants to stop a preview, although the camera had been started with the option ```show_preview=False```.

Probably, I need to get back to the Picamera2 team.

signag, 2024-05-08:
What you could do in the meantime:

When in the Settings screen, wait for more then 10 sec until the live stream stops (make sure that no other application is streaming).
Check by again pushing the *Settings* menu that the live stream indicator changed from red to gray.
Then, switching the camera will work.

Hermann-SW, 2024-05-08:
Thanks for the workaround — it worked.
Now cam1 is left and the one that got affected by zooming in.
[image]

signag, 2024-05-09:
Switching the camera should now work with V2.5.3 (See [Release Notes](https://github.com/signag/raspi-cam-srv/blob/main/docs/ReleaseNotes.md#v253))

Just to keep in mind:   
If you want to keep the zoom or other settings of the active camera after switching, you need to go to the *Web Cam* screen and push **Memorize Configuration and Controls for Camera Change** (see [User Guide](https://github.com/signag/raspi-cam-srv/blob/main/docs/Webcam.md#configuring-mjpeg-stream-and-jpeg-photo))

Hermann-SW, 2024-05-11:
> Switching the camera should now work with V2.5.3 (See [Release Notes](https://github.com/signag/raspi-cam-srv/blob/main/docs/ReleaseNotes.md#v253))
> 
It does, thank you!

> Just to keep in mind: If you want to keep the zoom or other settings of the active camera after switching, you need to go to the _Web Cam_ screen and push **Memorize Configuration and Controls for [...]
Thanks for that pointer, the memoize makes all work, and "Switch Cameras2 button below in "Web Cam" tab makes switching between cameras so easy!

I use my two GS cameras for high framerate video recording (>300fps). A typical resolution is 688x136@400fps. I used your streaming Web Cam view for adjustment, I set resolution to 688x136 both cameras in Config tab:
[image] 

Unfortunately that are not the cropped regions I get with high framerate GScrop tool:
https://gist.github.com/Hermann-SW/e6049fe1a24fc2b5a53c654e0e9f6b9c
https://stamm-wilbrandt.de/GS/
[image]
Captured with `GScrop 688 136 400 2000` and `cam1=1 GScrop 688 136 400 2000`

signag, 2024-05-13:
> Unfortunately that are not the cropped regions I get with high framerate GScrop tool:

I will have a look what the reasons might be.

> After those two commands: .... Starting server now does not work ...

Unfortunately, I cannot reproduce.
Were there any error messages?

Hermann-SW, 2024-05-13:
> Unfortunately, I cannot reproduce. Were there any error messages?
>
No fault on your side, its a libcamera thing.

I used your server and used webcam view seeing both cameras.
Then I stopped the server.
Then did `GScrop 688 136 400 200` seeing 2.5ms average frametime in matplotlib display.
Finally I started server again and reloaded webcam display.
Only the right camera works, left camera is not displayed and below errors happen.
```
[52:12:13.225802733] [31187] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video15[22:out]: Failed to start streaming: Invalid argument
[2024-05-14 00:36:48,074] ERROR in camera_pi: Thread 140734529466752:  CameraController.requestStart - Error starting camera: Failed to start camera: Invalid argument
[2024-05-14 00:36:48,075] ERROR in camera_pi: Thread 140734529466752: CameraController.requestCameraForConfig - camera could not be started
[2024-05-14 00:36:48,075] ERROR in camera_pi: Thread 140734529466752: Camera.frames - Exception: CameraController.requestCameraForConfig - Camera could not be started
[2024-05-14 00:36:48,075] ERROR in camera_pi: Thread 140734529466752: Camera._thread - Exception: CameraController.requestCameraForConfig - Camera could not be started
```

As I said, no prblem to get it working again.
I stopped server, executed `GScrop 1456 1088 60 200`, started server and now both cameras show in webcam view.

signag, 2024-05-18:
> Unfortunately that are not the cropped regions I get with high framerate GScrop tool:

I had a look at this and found that my knowledge of cropping in Picamera2/libcamera was quite poor.    
Unfortunately also the existing documentation could be more explicative.   
So, I did some systematic experiments, the results of which I've put in the chapter [Image Cropping and Sensor Modes](https://github.com/signag/raspi-cam-srv/blob/main/docs/ScalerCrop.md).
In [V2.6](https://github.com/signag/raspi-cam-srv/blob/main/docs/ReleaseNotes.md#v260) I had to completely revise the Zoom and Pan functionality.

Now, you should be able to better handle the 688x136 resolution.

The new button **Default** will set the ScalerCrop window to ScalerCrop Default which is what Picamera2 and the underlying libcamera will show if you do not explicitely set ScalerCrop.

Hermann-SW, 2024-05-19:
Thanks for the detailed documentation.
I will test the raspi-cam-srv 688x136 crop region when back from vacation.
This is first vacation since many years [without having a Pi with me](https://forum.prusa3d.com/forum/english-forum-awesome-prints-hall-of-fame/laptop-stand-for-very-high-display/#post-711303) ...

Hermann-SW, 2024-06-22:
Sorry, I am late.
And forgot my password.
Changing password is only possible knowing the old.
There is no reset password option in web login screen.
Is there a file I can delete that allows me for initial login?

signag, 2024-06-22:
The hard way would be to delete the database:
~/prg/raspi-cam-srv/instance/raspiCamSrv.sqlite

If you do not have any photo series, for which data are stored in the database, you will loose nothing but the user.

Hermann-SW, 2024-06-22:
Thanks, that worked.

I used raspi-cam-srv without issues after that, then stopped it.

Then I used my GScrop tool twice to crop to 688x136 for both cameras:
```
GScrop 688 136 402 500
cam1=1 GScrop 688 136 402 500
```

This is the result:
```
pi@raspberrypi5:~/jscadui/node_modules/@jscad/modeling/src $ rpicam-vid --list-cameras
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [406.67 fps - (376, 476)/688x136 crop]

1 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [406.67 fps - (376, 476)/688x136 crop]

pi@raspberrypi5:~/jscadui/node_modules/@jscad/modeling/src $ 
```

Then I started raspi-cam-srv again and accessed 127.0.0.1:5000 in browser.
No display, but an error message at bottom:
```
Error in Camera._thread: Error in live view: CameraController.requestCameraForConfig - Camera could not be started
Probably, a different camera configuration can solve the problem.
```

This behavior is same as when I try to use rpicam-vid in that situation.
To get rpicam-vid working again just execute "GScrop 1456 1088 60 100".
 
You said raspi-cam-srv should now work.
Can you please try yourself and use the GScrop command that matches (assuming you have only one GS camer connected).
Then you will see hopefully the error message I showed and can guide me how to resolve that.

This is the tool:
https://gist.github.com/Hermann-SW/e6049fe1a24fc2b5a53c654e0e9f6b9c

signag, 2024-06-25:
Hi,
I tried to use your tool and this is the result with just one GS camera connected (the only one I have):

```
(.venv) sn@raspi06:~/prg/GScrop $ ./GScrop 688 136 402 500
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

[0:48:44.739719068] [3358]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:48:44.753302914] [3361]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 14-02-2024 (14:00:12)
[0:48:44.770564208] [3361]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx296@1a to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:48:44.771472561] [3358]  INFO Camera camera.cpp:1183 configuring streams: (0) 688x136-YUV420
[0:48:44.771627950] [3361]  INFO RPI pisp.cpp:1405 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx296@1a - Selected sensor format: 1456x1088-SBGGR10_1X10 - Selected CFE format: 1456x1088-PC1B
[libx264 @ 0x555591f7e4a0] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x555591f7e4a0] profile High, level 3.2, 4:2:0, 8-bit
[libx264 @ 0x555591f7e4a0] 264 - core 164 r3095 baee400 - H.264/MPEG-4 AVC codec - Copyleft 2003-2022 - http://www.videolan.org/x264.html - options: cabac=1 ref=1 deblock=1:0:0 analyse=0x3:0x3 me=dia subme=0 psy=1 psy_rd=1.00:0.00 mixed_ref=0 me_range=16 chroma_me=0 trellis=0 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=0 threads=4 lookahead_threads=1 sliced_threads=0 slices=1 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=1 b_pyramid=0 b_adapt=1 b_bias=0 direct=1 weightb=0 open_gop=0 weightp=0 keyint=402 keyint_min=40 scenecut=0 intra_refresh=0 rc=crf mbtree=0 crf=23.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 pb_ratio=1.30 aq=1:1.00
Output #0, mp4, to '/dev/shm/tst.mp4':
  Stream #0:0: Video: h264, yuv420p(tv, smpte170m/smpte170m/bt709), 688x136, q=2-31, 402 fps, 402 tbr, 1000k tbn
Halting: reached timeout of 500 milliseconds.
[libx264 @ 0x555591f7e4a0] frame I:1     Avg QP:18.37  size: 15570
[libx264 @ 0x555591f7e4a0] frame P:15    Avg QP:19.91  size:  1452
[libx264 @ 0x555591f7e4a0] frame B:5     Avg QP:22.93  size:    77
[libx264 @ 0x555591f7e4a0] consecutive B-frames: 52.4% 47.6%
[libx264 @ 0x555591f7e4a0] mb I  I16..4: 13.2% 40.8% 46.0%
[libx264 @ 0x555591f7e4a0] mb P  I16..4:  7.7%  4.1%  0.2%  P16..4: 46.7%  0.0%  0.0%  0.0%  0.0%    skip:41.4%
[libx264 @ 0x555591f7e4a0] mb B  I16..4:  0.1%  0.0%  0.0%  B16..8:  3.0%  0.0%  0.0%  direct: 3.9%  skip:92.9%  L0:23.7% L1:76.3% BI: 0.0%
[libx264 @ 0x555591f7e4a0] 8x8 transform intra:36.6% inter:46.0%
[libx264 @ 0x555591f7e4a0] coded y,uvDC,uvAC intra: 53.1% 59.3% 27.2% inter: 14.3% 18.1% 3.7%
[libx264 @ 0x555591f7e4a0] i16 v,h,dc,p: 69% 11% 13%  7%
[libx264 @ 0x555591f7e4a0] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 26% 15% 29%  5%  3%  2%  3%  8%  8%
[libx264 @ 0x555591f7e4a0] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 34% 11% 16%  5%  8%  4%  5%  9%  9%
[libx264 @ 0x555591f7e4a0] i8c dc,h,v,p: 44% 21% 27%  8%
[libx264 @ 0x555591f7e4a0] kb/s:867.87

Total: 21 frames (20 samples)
Average: 16.563 ms / 60.376 fps
Minimum: 16.562 ms at frame 8
Maximum: 16.564 ms at frame 7
Outliers: 0 (100%) 0 (10.0%) 0 (1.0%) 0 (0.1%)
(.venv) sn@raspi06:~/prg/GScrop $ rpicam-vid --list-cameras
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

(.venv) sn@raspi06:~/prg/GScrop $
```

It seems that the tool did not change the camera settings.
Also starting raspiCamSrv afterwards does not produce an error

Anything, I am doing wrong with your tool?

Anyway, it seems that you are assuming that raspiCamSrv would work with libcamera settings which you have modified with GScrop before starting raspiCamSrv.
I would not expect that this could work.
raspiCamSrv is based on Picamera2, which itself uses libcamera.
When Picamera2 is instantiated, it exclusively initializes the camara using libcamera.
Probably, your error originates in this step if Picamera2 recognizes that within libcamera the camera is not within one of its allowed sensor modes.

Maybe, the Picamera2 team could make modifications which allow this.
But that's just guessing.
In any case, we would need the detailed error.

Since in my system, your tool does not seem to be changing the camera setting, you could probably activate debug logging in raspiCamSrv.
This is described in [Logging](https://github.com/signag/raspi-cam-srv/blob/main/docs/Troubelshooting.md#logging)

Hermann-SW, 2024-06-25:
Hi, do you use latest GScrop gist?
There should be a "/dev/media..." line first.
And you should get nearly 200 frames captured.
And the final list-cameras should show modified mode as below.
 What does "which media-ctl" output on your system?
```
pi@raspberrypi5:~ $ which media-ctl 
/usr/bin/media-ctl
pi@raspberrypi5:~ $ 
```

```
pi@raspberrypi5:~ $ GScrop 688 136 402 500
/dev/media0

Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [406.67 fps - (376, 476)/688x136 crop]

1 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

[105:55:35.087187103] [59859]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[105:55:35.095771010] [59862]  INFO RPI pisp.cpp:695 libpisp version v1.0.6 b567f0455680 17-06-2024 (10:20:00)
[105:55:35.113344901] [59862]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a to CFE device /dev/media0 and ISP device /dev/media3 using PiSP variant BCM2712_C0
[105:55:35.113409957] [59862]  INFO RPI pisp.cpp:695 libpisp version v1.0.6 b567f0455680 17-06-2024 (10:20:00)
[105:55:35.122339329] [59862]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx296@1a to CFE device /dev/media1 and ISP device /dev/media4 using PiSP variant BCM2712_C0
[105:55:35.122986611] [59859]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
[105:55:35.123186686] [59859]  INFO Camera camera.cpp:1183 configuring streams: (0) 688x136-YUV420
[105:55:35.123248761] [59862]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a - Selected sensor format: 688x136-SBGGR10_1X10 - Selected CFE format: 688x136-PC1B
[libx264 @ 0x5555c1e5d510] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x5555c1e5d510] profile High, level 3.2, 4:2:0, 8-bit
[libx264 @ 0x5555c1e5d510] 264 - core 164 r3095 baee400 - H.264/MPEG-4 AVC codec - Copyleft 2003-2022 - http://www.videolan.org/x264.html - options: cabac=1 ref=1 deblock=1:0:0 analyse=0x3:0x3 me=dia subme=0 psy=1 psy_rd=1.00:0.00 mixed_ref=0 me_range=16 chroma_me=0 trellis=0 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=0 threads=4 lookahead_threads=1 sliced_threads=0 slices=1 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=1 b_pyramid=0 b_adapt=1 b_bias=0 direct=1 weightb=0 open_gop=0 weightp=0 keyint=402 keyint_min=40 scenecut=0 intra_refresh=0 rc=crf mbtree=0 crf=23.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 pb_ratio=1.30 aq=1:1.00
Output #0, mp4, to '/dev/shm/tst.mp4':
  Stream #0:0: Video: h264, yuv420p(tv, smpte170m/smpte170m/bt709), 688x136, q=2-31, 402 fps, 402 tbr, 1000k tbn
Halting: reached timeout of 500 milliseconds.
[libx264 @ 0x5555c1e5d510] frame I:1     Avg QP:14.53  size:  5272
[libx264 @ 0x5555c1e5d510] frame P:94    Avg QP:21.73  size:  1128
[libx264 @ 0x5555c1e5d510] frame B:93    Avg QP:24.00  size:   374
[libx264 @ 0x5555c1e5d510] consecutive B-frames:  1.1% 98.9%
[libx264 @ 0x5555c1e5d510] mb I  I16..4: 17.8% 62.3% 19.9%
[libx264 @ 0x5555c1e5d510] mb P  I16..4: 34.8% 33.7%  0.1%  P16..4: 16.6%  0.0%  0.0%  0.0%  0.0%    skip:14.9%
[libx264 @ 0x5555c1e5d510] mb B  I16..4:  8.7%  5.1%  0.0%  B16..8: 18.2%  0.0%  0.0%  direct: 6.7%  skip:61.5%  L0:46.8% L1:48.7% BI: 4.5%
[libx264 @ 0x5555c1e5d510] 8x8 transform intra:47.3% inter:64.9%
[libx264 @ 0x5555c1e5d510] coded y,uvDC,uvAC intra: 31.6% 20.3% 1.8% inter: 3.9% 11.7% 0.0%
[libx264 @ 0x5555c1e5d510] i16 v,h,dc,p: 13% 74% 10%  3%
[libx264 @ 0x5555c1e5d510] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu:  6% 57% 22%  2%  2%  1%  3%  2%  5%
[libx264 @ 0x5555c1e5d510] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 13% 33% 27%  3%  4%  2%  4%  5%  9%
[libx264 @ 0x5555c1e5d510] i8c dc,h,v,p: 64% 25% 10%  2%
[libx264 @ 0x5555c1e5d510] kb/s:2513.16

Total: 188 frames (187 samples)
Average: 2.474 ms / 404.187 fps
Minimum: 2.468 ms at frame 27
Maximum: 2.480 ms at frame 163
Outliers: 0 (100%) 0 (10.0%) 0 (1.0%) 20 (0.1%)
pi@raspberrypi5:~ $ rpicam-vid --list-cameras
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [406.67 fps - (376, 476)/688x136 crop]

1 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

pi@raspberrypi5:~ $
```

signag, 2024-06-25:
```
sn@raspi06:~ $ which media-ctl
/usr/bin/media-ctl
```

```
sn@raspi06:~ $ media-ctl -d "/dev/media0"
sn@raspi06:~ $ media-ctl -d "/dev/media1"
sn@raspi06:~ $ media-ctl -d "/dev/media2"
sn@raspi06:~ $ media-ctl -d "/dev/media3"
sn@raspi06:~ $ media-ctl -d "/dev/media4"
Failed to enumerate /dev/media4 (-2)
sn@raspi06:~ $ media-ctl -d "/dev/media5"
Failed to enumerate /dev/media5 (-2)
sn@raspi06:~ $```

signag, 2024-06-25:
> I learned how to use picamera2 for dual camera display yesterday.
> But that only works locally and not in the browser as your raspi-cam-srv:

In principle, the raspiCamSrv Live stream, which is also used in the Web Cam screen, uses logically the same code except that I do not use the QT-preview, which only works with connected display, but streaming the lores or main stream, depending on configuration.

But it's interesting: If you ran your GScrop tool before starting Picamera2, then this should also work with raspiCamSrv.
It would, therefore be interesting to analyze the error you get.

signag, 2024-06-26:
I have modified your code to

```
#!/bin/bash
# shellcheck disable=SC2154
# (silence shellcheck wrt $cam1 environment variable)
if [[ $# -lt 4 ]];  then  echo "Format: [narrow=1] [cam1=1] $0 width height framerate ms [us]"; exit;  fi
if [[ "$(( $1 % 2 ))" -eq 1 ]];  then echo "width has to be even"; exit;  fi
if [[ "$(( $2 % 2 ))" -eq 1 ]];  then echo "height has to be even"; exit;  fi
export SHTR=""; if [[ $# -gt 4 ]]; then SHTR="--shutter"; fi
export workaround=""; if [[ "" != "$(grep '=bookworm' /etc/os-release)" ]]; then workaround="--no-raw"; fi
export d=10; if [[ "" != "$(grep "Revision.*: ...17.$" /proc/cpuinfo)" ]]; then if [[ "$cam1" == "" ]]; then d=6; else d=4; fi; fi
for((m=0; m<=5; ++m))
do
    echo media-ctl -d "/dev/media$m" --set-v4l2 "'imx296 $d-001a':0 [fmt:SBGGR10_1X10/${1}x$2 crop:($(( (1440 - $1) / 2 )),$(( (1088 - $2) / 2 )))/${1}x$2]"
    media-ctl -d "/dev/media$m" --set-v4l2 "'imx296 $d-001a':0 [fmt:SBGGR10_1X10/${1}x$2 crop:($(( (1440 - $1) / 2 )),$(( (1088 - $2) / 2 )))/${1}x$2]"
    #if  media-ctl -d "/dev/media$m" --set-v4l2 "'imx296 $d-001a':0 [fmt:SBGGR10_1X10/${1}x$2 crop:($(( (1440 - $1) / 2 )),$(( (1088 - $2) / 2 )))/${1}x$2]" >/dev/null;  then  echo -e "/dev/media$m\n"; break;  fi
done
```

and get

```
(.venv) sn@raspi06:~/prg/GScrop $ ./GScrop 688 136 402 500
media-ctl -d /dev/media0 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media1 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media2 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media3 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media4 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Failed to enumerate /dev/media4 (-2)
media-ctl -d /dev/media5 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Failed to enumerate /dev/media5 (-2)
(.venv) sn@raspi06:~/prg/GScrop $
(.venv) sn@raspi06:~/prg/GScrop $ media-ctl -d /dev/media0 --set-v4l2 "'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]" -v
Opening media device /dev/media0
Enumerating entities
looking up device: 81:0
looking up device: 81:1
looking up device: 81:2
looking up device: 81:3
looking up device: 81:4
looking up device: 81:5
looking up device: 81:6
looking up device: 81:7
looking up device: 81:8
Found 10 entities
Enumerating pads and links
no such entity "imx296 6-001a"

 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
  ^
Unable to parse format
Unable to setup formats: Invalid argument (22)
(.venv) sn@raspi06:~/prg/GScrop $
```

What is wrong?

Just to mention:
```
(.venv) sn@raspi06:~/prg/GScrop $ rpicam-vid --list-cameras
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]
(.venv) sn@raspi06:~/prg/GScrop $ cat /etc/debian_version
12.5
(.venv) sn@raspi06:~/prg/GScrop $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
(.venv) sn@raspi06:~/prg/GScrop $ uname -r
6.1.65-v8-16k+
(.venv) sn@raspi06:~/prg/GScrop $
```

Hermann-SW, 2024-06-26:
Hi, when I run your modified shell script I get this:
```
pi@raspberrypi5:~ $ ./GScrop.mod 688 136 402 500
media-ctl -d /dev/media0 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
media-ctl -d /dev/media1 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media2 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media3 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media4 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media5 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Failed to enumerate /dev/media5 (-2)
pi@raspberrypi5:~ $ 
```

When I run my GScrop it reports /dev/media0 first:
```
pi@raspberrypi5:~ $ GScrop 688 136 402 500
/dev/media0

Available cameras
...
```

In my output of your script /dev/media0 shows no error.
The code determines which /dev/mediaX device camera is connected to, as that can chnage on every reboot.

In your output all /dev/mediaX are bad, so that does not work.
I have the same OS output as you.
But you run a pretty old kernel (6.1.65), I run latest:
```
$ uname -r
6.6.31+rpt-rpi-2712
$ 
```

Can you please try on a fully updated SD card or SSD?
RPI engineers had fixed many libcamera issues, likely libcamera on your system does not work correctly with media-ctl, which GScrop tool relies on.

signag, 2024-06-27:
I have prepared a fresh SD card with Imager 1.8.5 (current version) and fully upgraded the system.

Same result:

```
(.venv) sn@raspi06:~/prg/GScrop $ ./GScrop 688 136 402 500
media-ctl -d /dev/media0 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media1 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media2 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media3 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media4 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Failed to enumerate /dev/media4 (-2)
media-ctl -d /dev/media5 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Failed to enumerate /dev/media5 (-2)
(.venv) sn@raspi06:~/prg/GScrop $ rpicam-vid --list-cameras
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SBGGR10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

(.venv) sn@raspi06:~/prg/GScrop $ uname -r
6.6.20+rpt-rpi-2712
(.venv) sn@raspi06:~/prg/GScrop $
```

The kernel (6.6.20) is still behind yours (6.6.31).
Strange, but I do not think that this could cause the discrepancy.

Can it be that you are using some secret configurations? 
Perhaps something with the legacy camera stack, which is, however, deprecated and no longer available in bookworm raspi-config?

signag, 2024-06-27:
Ah, I remembered why my kernel version was so old.
Had mentioned it in the [Trouble Shooting Guide](https://github.com/signag/raspi-cam-srv/blob/main/docs/Troubelshooting.md)
In [libcamera Issue #104](https://github.com/raspberrypi/libcamera/issues/104) it had been recommended to avoid a libcamera failure on Pi5 by going back to kernel version 6.1.65.

A post from naushir at Mar 8 states that the latest kernel version include the fix.

signag, 2024-06-27:
Now updated to the latest kernel version 6.6.35.

Same result, now with a V3 and a GS camera connected:

```
sn@raspi06:~/prg/GScrop $ ./GScrop 688 136 402 500
media-ctl -d /dev/media0 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media1 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media2 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media3 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media4 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Unable to setup formats: Invalid argument (22)
media-ctl -d /dev/media5 --set-v4l2 'imx296 6-001a':0 [fmt:SBGGR10_1X10/688x136 crop:(376,476)/688x136]
Failed to enumerate /dev/media5 (-2)
sn@raspi06:~/prg/GScrop $ rpicam-vid --list-cameras
Available cameras
-----------------
0 : imx708 [4608x2592 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

1 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

sn@raspi06:~/prg/GScrop $ uname -r
6.6.35-v8-16k+
sn@raspi06:~/prg/GScrop $
```

Hermann-SW, 2024-06-27:
Thanks for the updates.
GScrop tool only works for GS cameras.
Your GS camera is connected as cam1, and you need to tell that to GScrop,
```
$ GScrop 
Format: [narrow=1] [cam1=1] /usr/local/bin/GScrop width height framerate ms [us]
$
```
Please run this command, and you should see nearly 800 frames getting recorded in the 2s recording time:
```
pi@raspberrypi5:~/picamera2/examples $ cam1=1 GScrop 688 136 402 2000
/dev/media1

Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]

1 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [406.67 fps - (376, 476)/688x136 crop]

[157:56:04.126218598] [87172]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
...
[libx264 @ 0x555639d4d780] kb/s:1385.80

Total: 795 frames (794 samples)
Average: 2.474 ms / 404.186 fps
Minimum: 2.465 ms at frame 285
Maximum: 2.481 ms at frame 421
Outliers: 0 (100%) 0 (10.0%) 0 (1.0%) 59 (0.1%)
pi@raspberrypi5:~/picamera2/examples $ 
```

Now a really nice bonus — if you have installed matplotlib in your virtual environment (GScrop uses  ```~/venv/bin/python``` to start ```~/rpicam-apps/utils/timestamp.py```), you will get this nice analysis of the recorded .mp4 video timestamps! After closing that window the GScrop command will complete:
[image]

signag, 2024-06-28:
OK, I finally got it working.

I have now analyzed the error.
Using the [Code Generation](https://github.com/signag/raspi-cam-srv/blob/main/docs/Troubelshooting.md#generation-of-python-code-for-camera) feature of raspiCamSrv, I recorded the Picamera2 code which is applied by raspiCamSrv when the live stream is started.

I then simplified this code by removing camera configuration and setting of controls to get

```
from picamera2 import Picamera2
from picamera2.encoders import MJPEGEncoder
from picamera2.outputs import FileOutput
import time
import os
Picamera2.set_logging(Picamera2.ERROR)
os.environ["LIBCAMERA_LOG_LEVELS"] = "*:3"
videoDuration = 10
picam2 = Picamera2(0)
picam2.start(show_preview=False)
time.sleep(1.5)
output = None
encoder = MJPEGEncoder()
picam2.start_encoder(encoder, FileOutput(output))
time.sleep(videoDuration)
```

If I run this Python code after your cropping has been applied, I get a libcamera error:

```[1:38:39.843343255] [5075] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video4[19:cap]: Failed to start streaming: Invalid argument```

That is the reason why raspiCamSrv fails starting the camera.

If the code is run after cropping has been set back to standard, no error occurs.

Need to investigate this further.