signag/raspi-cam-srv #15: Video_Feed image size

katie1348, 2024-05-10:
Hi,

This might not be a problem, only user error, but I don't know where else to get the information I might need.

I have setup the latest version as of today and I can stream from 2 cameras (PiCam 3) at the same time on my Raspberry Pi 5 4GB. I can change the resolution of one of them. But I cannot see how to change the resolution of the other and I cannot seem to get the settings to save.

Is there documentation I am missing?

Your help is appreciated.

signag, 2024-05-10:
Hi,
thanks for your post.

Probably, the related chapter in the [User Guide](https://github.com/signag/raspi-cam-srv/blob/main/docs/UserGuide.md) might be what you are looking for: [Configuring MJPEG Stream and jpeg Photo](https://github.com/signag/raspi-cam-srv/blob/main/docs/Webcam.md#configuring-mjpeg-stream-and-jpeg-photo).

Let me know if you are missing something.

katie1348, 2024-05-10:
@signag Thank you for that help, it gave me exactly what I wanted and needed. My cameras are now setup and running at a good resolution and streaming happily to both endpoints.

You have done a good job with the software and it has really helped my. The setup instructions were easy to follow and get it working.

The only thing I would add, is perhaps at the end of the setup, you provide a link to the docs. I didn't think to go into the code tree looking for instructions. My bad, but that is what happened.

All working now and please close this when you have read the comment.

closed 2024-05-10 by signag