signag/raspi-cam-srv #19: Changing Video from Sensor Mode 0 to Sensor Mode 1 breaks the server

Tatterz719, 2024-06-27:
Pi 4 with Camera Module 3

https://youtu.be/5rqW24X3JKc?feature=shared

signag, 2024-06-30:
> I also still am unable to have a Live Preview at the same time as Recording.

With the default configurations for Pi4, I see the live stream while video is recording.
The same if the live view uses Sensor Mode 0:
[image]

[image]

[image]

[image]

signag, 2024-06-30:
> Pi 4 with Camera Module 3
> 
> https://youtu.be/5rqW24X3JKc?feature=shared

After changing sensor mode to 1 for video, the error is also shown on the status line at the bottom:

[image]

This is an error in the encoder which is propagated by raspiCamSrv.

Still need to check why the server reset does not work.

tatterz91, 2024-07-03:
So the project only functions correctly on Sensor Mode 0? And even when I use Sensor Mode 0, it shows the Preview while recording but the problem is that the resulting video file is still low frame rate like Preview shows. Getting rid of the preview gives me a usable video however.

[image]
Also, looks like its happening in your reply but I updated to Bookworm and am now getting an issue with the Noise of the preview

signag, 2025-02-10:
Hi

after all the time and new program version, I would be interested to hear whether your problems with the Pi 4 still exist.

Thanks

neretin, 2025-05-03:
> Hi
> 
> after all the time and new program version, I would be interested to hear whether your problems with the Pi 4 still exist.
> 
> Thanks

Can confirm, the issue still exists. 
Raspberry Pi 4 Model B Rev 1.5
Camera imx708

Thanks for a cool project!