signag/raspi-cam-srv #22: Web Cam Photo URL returns invalid picture when saved

lefake, 2024-07-19:
I'm trying to intergrate my rpi-cam with HomeAssistant and trying to trigger a picture when my motion sensor detect something (I know I could do something similar only with raspi-cam-srv but I already have some motion sensor so...).

If I use the URL given in the WebCam tab I can see the image in my browser but if I save it (right click -> save as) and open it with my image viewer I get this error : 
`Format error decoding Jpeg: Invalid JPEG format: first two bytes are not an SOI marker`

And I'm getting the same issue when setting the Picture URL into my HA setup.

Taking picture in the Live tab and saving them from the Photos tabs works just fine tho... 

Happy to provide any information you need !

signag, 2024-07-19:
Thanks for the report.

This is fixed in [V2.7.1](https://github.com/signag/raspi-cam-srv/blob/main/docs/ReleaseNotes.md#v271)

lefake, 2024-07-19:
Great this fixes all of it ! 
Thanks for the quick update !

closed 2024-07-19 by lefake