signag/raspi-cam-srv #23: Increasing the quality of live view

moataz-youssef, 2024-08-01:
First of all thanks for the awesome work. I have previously been using RPi-Cam-Web-Interface (https://github.com/silvanmelchior/RPi_Cam_Web_Interface/). There was the live stream quality a lot better. Is there anything I can do to increase the quality? I can go down with the resolution but I don't want to have visible artifacts or pixelation while streaming the video. Thanks a lot in advance.

moataz-youssef, 2024-08-01:
I am running raspi-cam-srv on a RPi 4B on the HQ Camera and the system is connected to a PC using gigabit ethernet.

moataz-youssef, 2024-08-02:
> I also see the bad quality of the live stream for the Pi4. But I do not see this on a Pi2 or a Pi5, on which I am testing, too. Need to investigate what is specific for the Pi4.

Thanks for your reply. Do you mean PiZero 2? 

So far I have tested your software with a Pi4 and a PiZero 1W and I have to say that the streaming quality was not satisfactory and changing the settings usually results in instability where I have to restart the server.

I have a Pi5 and a PiZero 2, so I might give it a go. But if the quality can be improved on a Pi4, this will be awesome.

signag, 2024-08-02:
Below are screenshots for Pi Zero, Pi Zero 2, Pi 4 and Pi 5, all with different cameras but with the default configurations:

Pi Zero W:

[image]

[image]

Pi Zero 2 W:

[image]

[image]

Pi 4:

[image]

[image]

Pi 5:

[image]

[image]

It can be seen that only the Pi 4 shows a bad quality of the preview.

Actually, raspiCamSrv just starts the camera in a specific mode and does not really have influence on the preview quality, except for controlling the camera configuration and controls to be used.

signag, 2025-02-10:
Hi 

comming back to this issue from long ago ...

the bad quality on the live stream for the Pi 4 seems to be related to the YUV format which according to the [Picamera2 manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) must be used for the lores stream for Pi4 and earlier.

Changing the settings for the *Live View* to *Stream* "main" and *Stream Format* "RGB888" gives much better quality.

See also my [comment](https://github.com/signag/raspi-cam-srv/issues/37#issuecomment-2647896200) to issue #37.

Could we close this issue?

Thanks

moataz-youssef, 2025-02-11:
> Hi
>
> comming back to this issue from long ago ...
>
> the bad quality on the live stream for the Pi 4 seems to be related to the YUV format which according to the [Picamera2 [...]

Thanks for the update. I will give it a try. Is this specific to the RPi4/HQ Camera combo or just for the RPi4 regardless of the camera used?

signag, 2025-02-11:
As far as I can see, this applies to all cameras on a Pi 4.
On Pi Zero and Pi Zero 2, I use the same default settings as for the Pi 4 with *Stream* "lores" and *Stream Format* "YUV420" and there I see normal quality.

moataz-youssef, 2025-02-20:
I still see a quality difference between Stream and Photo. Moreover, when both are set to use the "main" format, taking a photo crashes the live stream. The live stream stops and no error is displayed. 

Here are my settings for live:

[image]

And here are those for photo:

[image]