signag/raspi-cam-srv #26: NoIR camera settings

mister-bogus, 2024-08-23:
How can I include the settings for my V2.1 NoIR camera in order to get rid of the red tint?

In picamera the code is like this:

tuning = Picamera2.load_tuning_file("imx219_noir.json")
picam2 = Picamera2(tuning=tuning)

Thank you!

signag, 2024-08-23:
Thanks for the hint.

I did not deal with tuning so far because I assumed that libcamera would automatically load the appropriate tuning files.

Let me see how to integrate explicit tuning support into raspiCamSrv.

signag, 2024-08-31:
With V2.10.0, raspiCamSrv supports handling of tuning files.

I hope this covers what you need.

If you are using a Pi5 with 2 cameras, you should regard the note in the [Release Notes](https://github.com/signag/raspi-cam-srv/blob/main/docs/ReleaseNotes.md) as well as the chapter [Tuning with Multiple Cameras](https://github.com/signag/raspi-cam-srv/blob/main/docs/Tuning.md#tuning-with-multiple-cameras)

mister-bogus, 2024-09-01:
Thanks a lot. Works great!

closed 2025-02-10 by signag