signag/raspi-cam-srv #35: Triggered capture stopped with error

istvanrath, 2024-12-23:
I get the following error when trying to configure any motion detection:

`
Error in motionDetector._motionThread: Error in motion detection: OpenCV(4.6.0) ./modules/imgproc/src/color.simd_helpers.hpp:92: error: (-2:Unspecified error) in function 'cv::impl::{anonymous}::CvtHelper<VScn, VDcn, VDepth, sizePolicy>::CvtHelper(cv::InputArray, cv::OutputArray, int) [with VScn = cv::impl::{anonymous}::Set<3, 4>; VDcn = cv::impl::{anonymous}::Set<1>; VDepth = cv::impl::{anonymous}::Set<0, 2, 5>; cv::impl::{anonymous}::SizePolicy sizePolicy = cv::impl::<unnamed>::NONE; cv::InputArray = const cv::_InputArray&; cv::OutputArray = const cv::_OutputArray&]' > Invalid number of channels in input image: > 'VScn::contains(scn)' > where > 'scn' is 1
`

[image]

signag, 2024-12-25:
Hi,
thanks for this post.
Could you give me any hints which could help to reproduce this, for example the type of motion detection you are using.
To be honest, in general I am seing myself more issues on the Pi4 than I see on the Pi2 or Pi5.
Probably, I need to have a closer look on the specific Picamera2 restrictions which apply for a Pi4.

istvanrath, 2024-12-26:
The issues appear on a fresh vanilla install of the latest Git master, as per following the procedure at https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#system-setup

The problem doesn't show with "Mean Square Diff", as I guess that doesn't use OpenCV.

From a bit of googling around, it seems that the code is passing a grayscale (scn=1) image to a function that expects an RGB (scn=3) image. As the picture I see in the live view is clearly in full color, this seems weird. 

Everytime I try setting another sensor mode, I get the following error in the syslog:
`
kernel: [110478.067245] bcm2835-codec bcm2835-codec: Failed setting ctrl 00990a67, ret -3
`

Sporadically, I also get error messages like:
`
[30:46:29.930595067] [233175] ERROR Camera camera.cpp:706 Camera in Available state trying configure() requiring state between Acquired and Configured
[2024-12-26 16:28:36,093] ERROR in camera_pi: Thread 546391912832: CameraController.requestStart - Error starting camera: Failed to configure camera: Permission denied
[30:46:29.933831009] [233175] ERROR Camera camera.cpp:706 Camera in Available state trying configure() requiring state between Acquired and Configured
[2024-12-26 16:28:36,096] ERROR in camera_pi: Thread 546391912832: CameraController.requestStart - Error starting camera: Failed to configure camera: Permission denied
`

Additionally, I get two error messages as follows:
`
[30:29:26.854429235] [230081] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[30:29:26.854491326] [230081] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
`
But I guess these are totally unrelated and most probably harmless: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3411

Other than the motion detection with OpenCV, everything else seems to be working well, including live view, capturing photos and videos, etc.

If there's anything more I can do to help you (e.g. activate a debug flag, etc), just LMK :)

signag, 2024-12-26:
Two of the algorithms are using intermediate grayscale images: "Frame Differencing" and "Optical Flow".
With which of these do you get the motion detection error? 

The other errors seem to occur in lower levels of teh camera stack.
I don't know about compatibility of OV cameras with specific Raspberry Pi kernel versions.

> Everytime I try setting another sensor mode, I get the following error in the syslog

Does this occur for all sensor modes or only for specific one?

> Sporadically, I also get error messages like:

As commented in a [similar error](https://github.com/raspberrypi/picamera2/issues/483), this might occur when trying to reopen a camera which is already open.
Normally, I systematically try to avoid this.
Therefore, it would be helpful if you could reproduce this and indicate the specific situation when it occurs.

> Additionally, I get two error messages as follows:

libcamera also uses the IPAModule. So, there might be an issue. But I have currently no idea.

I'll try to do some more testing with my Raspberry Pi cameras on a Pi 4 to see whether I can reproduce some of the issues.

istvanrath, 2024-12-26:
> Two of the algorithms are using intermediate grayscale images: "Frame Differencing" and "Optical Flow".
With which of these do you get the motion detection error?

I get it with Frame Differencing, Optical Flow as well as Background Subtraction. The error is shown _before_ any motion detection can take place, so my guess is that is happens very early in the process, around the time when OpenCV gets first initialized. The "Test Motion Detection" screen, for instance, doesn't show any images.

My kernel version is:
`Linux cutiepi 6.6.67-v8+ #1833 SMP PREEMPT Fri Dec 20 14:14:15 GMT 2024 aarch64 GNU/Linux`

I'm running fresh install of Raspberry OS 12.8 Bookworm.

> Does this occur for all sensor modes or only for specific one?

Pretty much all.

istvanrath, 2024-12-26:
Btw. this is a PiCamera 1, with the original OmniVision OV5647 sensor.

As an experiment, I also tried running the app with `libcamerify`:
```
$ libcamerify flask --app raspiCamSrv run --port 5000 --host=0.0.0.0
[36:24:23.287998870] [277553] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[36:24:23.288106054] [277553] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[36:24:23.288140424] [277553]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[36:24:23.317992786] [277559]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[36:24:23.320275332] [277559]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media0
 * Serving Flask app 'raspiCamSrv'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:5000
 * Running on http://192.168.0.109:5000
Press CTRL+C to quit
[36:24:36.183149334] [277702] FATAL Process process.cpp:117 Multiple ProcessManager objects are not allowed
Backtrace not available
Aborted
```

As you can see, it dies as soon as I try to connect to the web UI.

istvanrath, 2024-12-26:
It seems I managed to get it working by changing the Stream format to RGB888:

[image]

By default, it was set to YUV420.

signag, 2024-12-27:
I reopened it as a reminder to adjust the default for Pi4

signag, 2025-02-10:
Yes, changing the *Stream format* to RGB seems to be the solution.
However, according to the [Picamera2 manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) the format must be YUV for the lores stream for Pi4 and earlier.
See also my [comment](https://github.com/signag/raspi-cam-srv/issues/37#issuecomment-2647896200) to issue #37.

So, I'll prefer to keep the defaults as they are and close this issue.

closed 2025-02-10 by signag