signag/raspi-cam-srv #36: Rotate 90 degrees

thelostpatronus, 2024-12-23:
Thank you so much for this program yall, it is literally a lifesaver. Is there any way to rotate the live camera view 90 degrees? whether that is in the actual video embed or on the picamera2 level? Thanks so much :)

signag, 2024-12-25:
As far as I know, neither libcamera nor picamera nor picamera2 support rotations. 
Only mirroring is supported.

In [Picamera2 - How to rotate image 90, 180, 270 degrees?](https://forums.raspberrypi.com/viewtopic.php?t=364146) there are some discussions how rotation can be achieved for stored images and videos.
But this would probably not help for the live view.
I am currently not sure about EXIF data in MJPEG.

signag, 2025-02-10:
Hi,
in the Picamera2 examples, I found a solution for [rotation by 90 or 270 ° in an mjpeg server](https://github.com/raspberrypi/picamera2/blob/fb031cb49cb2af4dab7439be7a90784c9d3db1ab/examples/mjpeg_server_with_rotation.py#L17).
So, there should also be a possibility to implement that in raspiCamSrv.

I'll put it on the ToDo list, but probably for later.

dmelo, 2025-04-03:
The way I'm using the camera, because of how the cables are passing on my project, I have to keep the camera upside down. So it would be very useful to have the possibility to rotate 180. Any update on this issue?

dmelo, 2025-04-03:
An Update. Doing this worked for me. I'm too tired to continue right now. But I can try to make it configurable:

```
diff --git a/raspiCamSrv/camera_pi.py b/raspiCamSrv/camera_pi.py
index 3b6830c..75150ea 100644
--- a/raspiCamSrv/camera_pi.py
+++ b/raspiCamSrv/camera_pi.py
@@ -1476,6 +1476,8 @@ class Camera():
             # Initially set the stream size to (640, 480). Use Sensor Mode, if possible
             # If stream_size is set, keep the settings. They have been loeaded from stored config
             if cfg.liveViewConfig.stream_size is None:
+                cfg.liveViewConfig.transform_hflip = True
+                cfg.liveViewConfig.transform_vflip = True
                 sizeWidth = 640
                 sizeHeight =  int(sizeWidth * cfgProps.pixelArraySize[1] / cfgProps.pixelArraySize[0])
                 if (sizeHeight % 2) != 0:
```

signag, 2025-04-03:
Isn't that what you configure on the Config screen:

[image]

Changing the setting for one configuration will automatically transfer it to all others.

180° has never been a problem as it is for 90° and other angles. For these there is still no support in raspiCamSrv.

dmelo, 2025-04-04:
I missed this settings completely. Thanks a lot for pointing it out.