signag/raspi-cam-srv #37: changing camera from V2.1 to HQ camera - preview pixelated

pootle, 2025-01-10:
was running with Pi camera 2.1, everything fine.
switched camera for an HQ camera and display has gone 4 bit colour and pixelated.
have tried resetting in server settings and then reboot and changing live view config to different sizes but problem remains

photos are fine
[image]

signag, 2025-01-10:
Could  you post a screenshot of the Info screen showing your configuration?

pootle, 2025-01-10:
here you go. btw HQ camera worked perfectly on a pi 5 when first setup. This is on a pi 3B.
[image]

signag, 2025-01-10:
Could you please also show the Info screen, in addition to the Config screen?
Thanks

pootle, 2025-01-10:
Ah! I was wrong, it seems to be a problem with a Pi 3 vs Pi 5. I did a clean build and ran up a pi3 with the HQ cam and get the same pixelation problem. On a pi 5 it works properly.

pootle, 2025-02-11:
That's working on Pi3, excellent!

closed 2025-02-11 by pootle