signag/raspi-cam-srv #40: Package it in Docker / `docker-compose.yml`

GamePlayer-8, 2025-01-24:
Hello,

Any way that the project would have a `docker-compose.yml` for easy deployment?

signag, 2025-02-04:
Hi, 
thanks for this suggestion.

This is on my ToDo list since long.

GamePlayer-8, 2025-02-25:
Heya,

Have you tried adding:
```
    shm_size: '1gb'
    deploy:
      resources:
        limits:
          memory: '2g' # with zram if not enough memory
```

and see how it goes?

Also is `dmesg` printing anything special?

signag, 2025-02-28:
Hi
thanks for your suggestion.

However, the result is the same as before, except for the error message
```
! picamera2-test Your kernel does not support memory limit capabilities or the cgroup is not mounted. Limitation discarded.
```
Since ```docker info``` also reports
```
..
WARNING: No memory limit support
WARNING: No swap limit support
```
this is probably to be expected.

```dmesg``` sais, among others:
```
cma: cma_alloc: linux,cma: alloc failed, req-size: 729 pages, ret: -12
```

```cat /proc/meminfo``` shows:
```
...
CmaTotal:          65536 kB
CmaFree:           49056 kB
```
Since, according to the [picamera2-manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) ch. 8.3, the 12MP Model 3 image requires 48 MB, this will probably be the bottleneck.

I increased cma by replacing
```
dtoverlay=vc4-kms-v3d
```
with
```
dtoverlay=vc4-kms-v3d,cma-512
```
in ```/boot/firmware/config.txt```

Now, after reboot, I have:
```
CmaTotal:         524288 kB
CmaFree:          507808 kB
```
Now, the ```Cannot allocate memory``` error is gone when running the docker container with the Model 3 camera and a 12MP image is created.

This looks promising.
I'll continue trying

signag, 2025-03-07:
Hi,
if you like, you can try using Docker as described in [Running raspiCamSrv as Docker Container](https://github.com/signag/raspi-cam-srv/blob/main/docs/SetupDocker.md)

This has been added in V2.13.

Up to now, I have tested it on a Pi 5 and a Pi 4.