signag/raspi-cam-srv #41: Photo autfocus issue

mmib, 2025-01-31:
Raspberry Pi Zero 2 W Rev 1.0 
Board Revision 902120 
Kernel Version 6.6.62+rpt-rpi-v8 
Debian GNU/Linux 12 (bookworm) - Version 12.9

Arducam imx519 camera with focus.

Focus works fine on video however whatever i do I cannot get it work with photos. All photos are taken with LensPosition 1.0 (1.0 m), tried all possible focus handling combinations - it is ignored for photo but works fine for video

mmib, 2025-02-05:
Hi, thank you for reply. Repeated same steps, here is what i get (photo on screen is the old one):Error in Camera.takeImage: Phototaking caused error: Control AeEnable is not advertised by libcamera

[image]

mmib, 2025-02-05:
Server needs to be restarted after any error, otherwise there is no video stream. If would be good to implement some try-catch.

signag, 2025-02-05:
Hi, 
AeEnable refers to automatic exposure (see [picamera2_manual.pdf](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf), p. 79)
This control parameter does not seem to be supported by your camera. That's what the error message sais.
Up to now, I assume that all control parameters  documented in the  [picamera2_manual.pdf](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) are valid, except
- AfMode and all autofocus-related parameters
- AeFlickerMode
- HdrMode

For the latter, I check whether these parameters are available in the Picamera2.camera_controls structure.
This assumption applies to all original RaspberryPi cameras.
Probably, I would need to check for AeEnable too.

If you have activated any AutoExposure settings, you may try deactivating these.

However, this error is not related to focus handling.

signag, 2025-02-05:
> Server needs to be restarted after any error, otherwise there is no video stream. If would be good to implement some try-catch.

I know,
and I tried already to catch as much as I can.
But several of these errors occur in asynchronous threads started by Picamera and lower laying routines, which makes it difficult.