signag/raspi-cam-srv #44: Updated to latest version raspiCamSrv V2.10.4 (500 internal server error displays on several pages)

lgeitner, 2025-02-05:
Updated to latest version raspiCamSrv V2.10.4 (I was previously on version V2.5.3)

When I click on several pages I receive a 500 internal server error.

Info page displays following error:

 Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

sudo journalctl -e displays the following errors:  (Please note: I do not use tuning files!)

sudo journalctl -e
Feb 04 22:13:08 octopi flask[2101]:     tfl = getTuningFiles(tc.tuningFolder, tc.tuningFile)
Feb 04 22:13:08 octopi flask[2101]:           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:13:08 octopi flask[2101]:   File "/home/pi/camera0/raspi-cam-srv/raspiCamSrv/config.py", line 231, in getTuningFiles
Feb 04 22:13:08 octopi flask[2101]:     if os.path.exists(folder):
Feb 04 22:13:08 octopi flask[2101]:        ^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:13:08 octopi flask[2101]:   File "<frozen genericpath>", line 19, in exists
Feb 04 22:13:08 octopi flask[2101]: TypeError: stat: path should be string, bytes, os.PathLike or integer, not property
Feb 04 22:13:26 octopi sudo[2805]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/apt update
Feb 04 22:13:26 octopi sudo[2805]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Feb 04 22:13:28 octopi sudo[2805]: pam_unix(sudo:session): session closed for user root
Feb 04 22:13:37 octopi flask[2101]: [2025-02-04 22:13:37,173] ERROR in camera_pi: Thread 140732469604736: Camera.frames - Exception: expected str, bytes or os.PathLike object, not property
Feb 04 22:13:37 octopi flask[2101]: [2025-02-04 22:13:37,173] ERROR in camera_pi: Thread 140732469604736: Camera._thread - Exception: expected str, bytes or os.PathLike object, not property
Feb 04 22:13:37 octopi kernel: rp1-cfe 1f00128000.csi: Using a link rate of 900 Mbps
Feb 04 22:14:02 octopi flask[2101]: [2025-02-04 22:14:02,648] ERROR in app: Exception on /config [GET]
Feb 04 22:14:02 octopi flask[2101]: Traceback (most recent call last):
Feb 04 22:14:02 octopi flask[2101]:   File "/home/pi/camera0/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 1455, in wsgi_app
Feb 04 22:14:02 octopi flask[2101]:     response = self.full_dispatch_request()
Feb 04 22:14:02 octopi flask[2101]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:14:02 octopi flask[2101]:   File "/home/pi/camera0/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 869, in full_dispatch_request
Feb 04 22:14:02 octopi flask[2101]:     rv = self.handle_user_exception(e)
Feb 04 22:14:02 octopi flask[2101]:          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:14:02 octopi flask[2101]:   File "/home/pi/camera0/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 867, in full_dispatch_request
Feb 04 22:14:02 octopi flask[2101]:     rv = self.dispatch_request()
Feb 04 22:14:02 octopi flask[2101]:          ^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:14:02 octopi flask[2101]:   File "/home/pi/camera0/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 852, in dispatch_request
Feb 04 22:14:02 octopi flask[2101]:     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
Feb 04 22:14:02 octopi flask[2101]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:14:02 octopi flask[2101]:   File "/home/pi/camera0/raspi-cam-srv/raspiCamSrv/auth.py", line 273, in wrapped_view
Feb 04 22:14:02 octopi flask[2101]:     return view(**kwargs)
Feb 04 22:14:02 octopi flask[2101]:            ^^^^^^^^^^^^^^
Feb 04 22:14:02 octopi flask[2101]:   File "/home/pi/camera0/raspi-cam-srv/raspiCamSrv/config.py", line 52, in main
Feb 04 22:14:02 octopi flask[2101]:     tfl = getTuningFiles(tc.tuningFolder, tc.tuningFile)
Feb 04 22:14:02 octopi flask[2101]:           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:14:02 octopi flask[2101]:   File "/home/pi/camera0/raspi-cam-srv/raspiCamSrv/config.py", line 231, in getTuningFiles
Feb 04 22:14:02 octopi flask[2101]:     if os.path.exists(folder):
Feb 04 22:14:02 octopi flask[2101]:        ^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:14:02 octopi flask[2101]:   File "<frozen genericpath>", line 19, in exists
Feb 04 22:14:02 octopi flask[2101]: TypeError: stat: path should be string, bytes, os.PathLike or integer, not property
Feb 04 22:16:00 octopi sudo[3181]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/journalctl -e
Feb 04 22:16:00 octopi sudo[3181]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Feb 04 22:17:01 octopi CRON[3186]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Feb 04 22:17:01 octopi CRON[3187]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Feb 04 22:17:01 octopi CRON[3186]: pam_unix(cron:session): session closed for user root
Feb 04 22:17:15 octopi octoprint[798]: 2025-02-04 22:17:15,316 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Feb 04 22:18:42 octopi sudo[3181]: pam_unix(sudo:session): session closed for user root
Feb 04 22:18:51 octopi sudo[3198]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/systemctl restart raspiCamSrv0.service
Feb 04 22:18:51 octopi sudo[3198]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Feb 04 22:18:51 octopi systemd[1]: Stopping raspiCamSrv0.service - raspiCamSrv...
Feb 04 22:18:51 octopi systemd[1]: raspiCamSrv0.service: Deactivated successfully.
Feb 04 22:18:51 octopi systemd[1]: Stopped raspiCamSrv0.service - raspiCamSrv.
Feb 04 22:18:51 octopi systemd[1]: raspiCamSrv0.service: Consumed 2min 3.857s CPU time.
Feb 04 22:18:51 octopi systemd[1]: Started raspiCamSrv0.service - raspiCamSrv.
Feb 04 22:18:51 octopi sudo[3198]: pam_unix(sudo:session): session closed for user root
Feb 04 22:18:52 octopi flask[3202]:  * Serving Flask app 'raspiCamSrv'
Feb 04 22:18:52 octopi flask[3202]:  * Debug mode: off
Feb 04 22:18:52 octopi flask[3202]: WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
Feb 04 22:18:52 octopi flask[3202]:  * Running on all addresses (0.0.0.0)
[... 45 lines cut ...]
Feb 04 22:42:39 octopi flask[3427]: KeyError: 'tuningconfig'
Feb 04 22:42:40 octopi flask[3427]: [2025-02-04 22:42:40,554] ERROR in app: Exception on /info [GET]
Feb 04 22:42:40 octopi flask[3427]: Traceback (most recent call last):
Feb 04 22:42:40 octopi flask[3427]:   File "/home/pi/camera0/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 1455, in wsgi_app
Feb 04 22:42:40 octopi flask[3427]:     response = self.full_dispatch_request()
Feb 04 22:42:40 octopi flask[3427]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:42:40 octopi flask[3427]:   File "/home/pi/camera0/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 869, in full_dispatch_request
Feb 04 22:42:40 octopi flask[3427]:     rv = self.handle_user_exception(e)
Feb 04 22:42:40 octopi flask[3427]:          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:42:40 octopi flask[3427]:   File "/home/pi/camera0/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 867, in full_dispatch_request
Feb 04 22:42:40 octopi flask[3427]:     rv = self.dispatch_request()
Feb 04 22:42:40 octopi flask[3427]:          ^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:42:40 octopi flask[3427]:   File "/home/pi/camera0/raspi-cam-srv/.venv/lib/python3.11/site-packages/flask/app.py", line 852, in dispatch_request
Feb 04 22:42:40 octopi flask[3427]:     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
Feb 04 22:42:40 octopi flask[3427]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 04 22:42:40 octopi flask[3427]:   File "/home/pi/camera0/raspi-cam-srv/raspiCamSrv/auth.py", line 273, in wrapped_view
Feb 04 22:42:40 octopi flask[3427]:     return view(**kwargs)
Feb 04 22:42:40 octopi flask[3427]:            ^^^^^^^^^^^^^^
Feb 04 22:42:40 octopi flask[3427]:   File "/home/pi/camera0/raspi-cam-srv/raspiCamSrv/info.py", line 35, in main
Feb 04 22:42:40 octopi flask[3427]:     tcs[camnum] = cstrc["tuningconfig"]
Feb 04 22:42:40 octopi flask[3427]:                   ~~~~~^^^^^^^^^^^^^^^^
Feb 04 22:42:40 octopi flask[3427]: KeyError: 'tuningconfig'
Feb 04 22:43:14 octopi sudo[3607]: pam_unix(sudo:session): session closed for user root
Feb 04 22:43:52 octopi sudo[3709]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/journalctl -e
Feb 04 22:43:52 octopi sudo[3709]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)

signag, 2025-02-05:
Hi,
thanks for this post.

Just to clarify: is [Start Server with stored configuration](https://github.com/signag/raspi-cam-srv/blob/main/docs/Settings.md#server-configuration) activated?
This should be deactivated before upgrading (see [Update Procedure](https://github.com/signag/raspi-cam-srv/blob/main/docs/ReleaseNotes.md#update-procedure))

lgeitner, 2025-02-06:
I deleted the current raspi-cam-srv install, then manually reinstalled raspi-cam-srv with the instructions you provided.  Everything is working now.

Thank You for creating the raspiCamSrv web server!
lgeitner

closed 2025-02-06 by lgeitner