signag/raspi-cam-srv #45: No image/video attached to notification email

mmib, 2025-02-05:
Fixed email sending by commenting out passing username and password in code (would be good to skip this line if login or password is empty).
Now I getting motion detection emails, however no image and video attached. There is no video and image for such motion detentions in interface too.

signag, 2025-02-07:
Hi,

thanks for this post.

> (would be good to skip this line if login or password is empty).

This is covered in [V2.10.5](https://github.com/signag/raspi-cam-srv/blob/main/docs/ReleaseNotes.md#v2105).
You need to explicitely specify whether authentication is required and if so, you also need to specify user and password.

> Now I getting motion detection emails, however no image and video attached. There is no video and image for such motion detentions in interface too.

I have tested this again on a Pi5 with a Model 3 camera (imx708) and on a Pi Zero 2 W with a Model 2 camera (imx219)
Everything worked OK. Photos and Video were included in the mail.

Did you see event pictures and videos on the [Trigger - Events](https://github.com/signag/raspi-cam-srv/blob/main/docs/TriggerEventViewer.md#events) screen?

If not, make sure that *Take Photo* and *Record Video* on the [Trigger - Control](https://github.com/signag/raspi-cam-srv/blob/main/docs/Trigger.md) page is activated.

Can you take photos and videos in the Live view?

If you see photos and videos, make sure that *Include Video* and *Include Photos* is activated on the [Trigger - Notification](https://github.com/signag/raspi-cam-srv/blob/main/docs/TriggerNotification.md) screen.

If none of this helps, you may check for errors in the server log.
Eventually, you may activate [logging](https://github.com/signag/raspi-cam-srv/blob/main/docs/Troubelshooting.md#logging) for "raspiCamSrv.motionDetector" (line 72 in ```__init.py__```)

mmib, 2025-02-07:
> id you see event pictures and videos on the [Trigger - Events](https://github.com/signag/raspi-cam-srv/blob/main/docs/TriggerEventViewer.md#events) screen?
no
Record is enabled

[image]

Can take photos from the Live view, however cannot take videos: 
Error in Camera._videoThread: Error in encoder: [Errno 3] No such process
Probably, the requested resolution is too high.

this is what I'm getting in logs

`
Feb 07 11:54:18 rpizero flask[502]: [0:00:51.987227739] [927] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
Feb 07 11:54:18 rpizero flask[502]: [0:00:51.991204406] [927] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
Feb 07 11:54:21 rpizero flask[502]: [0:00:54.575814978] [932] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
Feb 07 11:54:21 rpizero flask[502]: [0:00:54.575935655] [932] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
Feb 07 11:54:23 rpizero kernel: bcm2835-codec bcm2835-codec: Failed setting ctrl 00990a67, ret -3
Feb 07 11:55:11 rpizero flask[502]: [0:01:44.747161795] [948] ERROR Camera camera.cpp:706 Camera in Available state trying configure() requiring state between Acquired and Configured
Feb 07 11:55:11 rpizero flask[502]: [2025-02-07 11:55:11,446] ERROR in camera_pi: Thread 547415781760: CameraController.requestStart - Error starting camera: Failed to configure camera: Permission denied
Feb 07 11:55:11 rpizero flask[502]: [2025-02-07 11:55:11,484] ERROR in camera_pi: Thread 547415781760: CameraController.requestCameraForConfig - camera could not be started
Feb 07 11:55:11 rpizero flask[502]: [2025-02-07 11:55:11,522] ERROR in camera_pi: Thread 547415781760: Camera.takeImage: Error CameraController.requestCameraForConfig - Camera could not be started
Feb 07 11:55:11 rpizero flask[502]: [0:01:45.011151223] [949] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
Feb 07 11:55:11 rpizero flask[502]: [0:01:45.011249867] [949] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
Feb 07 11:55:15 rpizero flask[502]: [2025-02-07 11:55:15,646] ERROR in camera_pi: Thread 547918508416: Camera._videoThread - Error: [Errno 3] No such process
Feb 07 11:55:15 rpizero kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -3
`

Seems need to play with my camera drivers. Thank you

signag, 2025-02-07:
If video recording causes the problem, you may try motion capturing with photos only to see whether this works.