signag/raspi-cam-srv #48: Frame Differencing motion detection test error:  Invalid number of channels in input image

jrbergen, 2025-03-15:
When pressing the Test Motion Detection button with the Frame Differencing Motion Detection algorithm selected, and the default settings, the following error shows up in the log when one tries to start the motion detection test:

ERROR in motionDetector: Exception in _motionThread: OpenCV(4.6.0) ./modules/imgproc/src/color.simd_helpers.hpp:92: error: (-2:Unspecified error) in function 'cv::impl::{anonymous}::CvtHelper<VScn, VDcn, VDepth, sizePolicy>::CvtHelper(cv::InputArray, cv::OutputArray, int) [with VScn = cv::impl::{anonymous}::Set<3, 4>; VDcn = cv::impl::{anonymous}::Set<1>; VDepth = cv::impl::{anonymous}::Set<0, 2, 5>; cv::impl::{anonymous}::SizePolicy sizePolicy = cv::impl::<unnamed>::NONE; cv::InputArray = const cv::_InputArray&; cv::OutputArray = const cv::_OutputArray&]'
> Invalid number of channels in input image:
>     'VScn::contains(scn)'
> where
>     'scn' is 1

[image]

Subsequently, the test does not seem to proceed:
[image]

Hardware+software context:
- Raspberry Pi 3 model B v 1.2
- Rasbperry pi camera module 3
- Debian bookworm, aarch64
- Raspi-cam-srv version 3.2.0
Relevant python packages versions in venv:
  - OpenCV 4.6.0
  - Flask 3.1.0
  - Flask-JWT-Extended 4.7.1
  - matplotlib 3.10.1
  - numpy 1.24.2
  - [Full list of installed Python packages in venv](https://pastebin.com/JJNAq5mH))

I'm willing to help myself, but I don't know (yet) how to debug a flask app on a remote server.

dillionhuston, 2025-03-15:
Have you maybe tried messing with settings to see if the error continues? and also to debug. just pass debug=true in your app.run

jrbergen, 2025-03-15:
Thanks for your quick reply.

> Have you maybe tried messing with settings to see if the error continues? 

I have tried various settings, but there are quite a few combinations possible, so I'd like to first know which settings might cause this error, by checking exactly where in the Python code this exception is raised.

> . just pass debug=true in your app.run

Oh nice. Didn't know it was that simple. Thanks!

I have to call it a day for now, but I will have some time to check again tomorrow; I'll try the remote debugging.

dillionhuston, 2025-03-15:
> Thanks for your quick reply.
> 
> > Have you maybe tried messing with settings to see if the error continues? 
> 
> I have tried various settings, but there are quite a few combinations possible, so I'd like to first know which settings might cause this error, by checking exactly in where in the Python code the OpenCV call is made leading to this exception.
> 
> > . just pass debug=true in your app.run
> 
> Oh nice. Didn't know it was that simple. Thanks!
> 
> I have to call it a day for now, but I will have some time to check again tomorrow; I'll try the remote debugging.

By reading the error message(correct me if I'm wrong) but it's seems to be something to do with your colour settings. As the error is referring to the channel input for image and at the start it is referring to the colour. Maybe have a look at changing your colour space.

I would have to look at the code first but please correct me if I'm wrong

signag, 2025-03-16:
Hi, 
the same exception was reported in #35.

The problem seems to be the YUV format.
The issue is, however, that for Raspberry Pi 4 and earlier, the stream format needs to be YUV for the lores stream.

For motion detection, I use the configuration for the Live View.
So, you might try switching the Stream for that config to "main" and choose the RGB888 format.
Also, I would recommend using a low custom stream size such as 640 x 400 because otherwise you might run into performance issues on the Pi3.

Unfortunately, I do not have a Pi 3 for testing, but I successfully tested it on a Pi 4 with the configuration described above.

signag, 2025-03-16:
> I'm willing to help myself, but I don't know (yet) how to debug a flask app on a remote server.

Hi, I have added this issue to the [Trouble Shooting Guide](https://github.com/signag/raspi-cam-srv/blob/main/docs/Troubelshooting.md).

There, you will also find information on how to activate detailed [logging for raspiCamSrv](https://github.com/signag/raspi-cam-srv/blob/main/docs/Troubelshooting.md#logging), should this be required.

jrbergen, 2025-03-16:
Thanks both! This seems to have been the issue indeed. I did not even need to debug. Nice that you added that to the trouble shooting guide as well; this will make it easier to identify issues in the future.

With these settings for live view, it works:

[image]