signag/raspi-cam-srv #50: I2S microphone not detected

RainerMinixhofer, 2025-03-22:
I hooked up an I2C microphone to the RaspiCamSrv and got it running through PulseAudio using the "googlevoicehat-soundcard" module (see [Adafruit I2S MEMS Microphone Breakout](https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-test)).
I can record audio through PulseAudo as described in the link. However the RaspiCamSrv shows "No Microphone found". Maybe there are some specifics to the SoundCard detection routine in RaspiCamSrv, preventing detection of unusual microphones?
BR,
Rainer

signag, 2025-03-22:
Hi Rainer,

did you consider the chapter [Recording Audio along with Video](https://github.com/signag/raspi-cam-srv/blob/main/docs/Settings.md#recording-audio-along-with-video).

If you are running raspiCamSrv as service,, you need to run it as user unit and not as system unit.

The way how to install this, is described under [Service Configuration for Audio Support](https://github.com/signag/raspi-cam-srv?tab=readme-ov-file#service-configuration-for-audio-support)

Please let me know whether this has solved the issue.

Thanks
Siegfried

RainerMinixhofer, 2025-03-23:
Hi Siegfried,

yes I considered the chapter and the raspiCamSrv is running as user unit since installation. This is the output:

xxx@xxx:~ $ systemctl --user status raspiCamSrv.service
● raspiCamSrv.service - raspiCamSrv
     Loaded: loaded (/home/xxx/.config/systemd/user/raspiCamSrv.service; enabled; preset: enabled)
     Active: active (running) since Sat 2025-03-22 16:03:26 CET; 1 day 7h ago
   Main PID: 724 (flask)
      Tasks: 6 (limit: 8736)
        CPU: 40.434s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/raspiCamSrv.service
             └─724 /home/xxx/Projects/raspi-cam-srv/.venv/bin/python /home/xxx/Projects/raspi-cam-srv/.venv/bin/flask --app raspiCamSrv run --port 5000 --host=0.0.0.0

Mar 22 16:03:26 xxx systemd[685]: Started raspiCamSrv.service - raspiCamSrv.
Mar 22 16:03:37 xxx flask[724]:  * Serving Flask app 'raspiCamSrv'
Mar 22 16:03:37 xxx flask[724]:  * Debug mode: off
Mar 22 16:03:37 xxx flask[724]: WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
Mar 22 16:03:37 xxx flask[724]:  * Running on all addresses (0.0.0.0)
Mar 22 16:03:37 xxx flask[724]:  * Running on http://127.0.0.1:5000
Mar 22 16:03:37 xxx flask[724]:  * Running on http://xxx.xxx.xxx.xxx:5000
Mar 22 16:03:37 xxx flask[724]: Press CTRL+C to quit

BR,
Rainer

signag, 2025-03-26:
Hi Rainer,

raspiCamSrv uses the following procedure to find the microphone:

1. Ececute ```pactl -fjson list sources```
2. Execute ```pactl get-default-source```
3. Iterate the sources (first level) found in step 1 until a source is found where "name" is identical with the value found in step 2
4. Then, iterate the "ports" element of this source until a port is found where "type" is "Mic".
5. If such a port has been found, the element "mute" of the source element is checked to see whether the microphone is muted.
6. If such a port is not found, it is assumed that no microphone is connected.

Maybe, you can follow this procedure to see what is different in your environment.

Thanks