signag/raspi-cam-srv #53: Camera in Running state trying release() requiring state between Available and Configured

wb1016, 2025-04-04:
# what i did
### attached Pi HQ camera to Pi 4B

### configured camera via config section:
- colorspace Rec.709
- sensor mode 0
- stream main
- stream format YUV420

### pressed 'Photo' to take photo

# expected behavior
the server succeed to take photo and resumes operation

# actual behavior
- the server stops output-ing camera feed
- photo is not saved
- unable to use camera at all (video,stream,photo, raw)
- all problem will continue until i restart the server

## log output
`[0:02:18.975520820] [1024] ERROR Camera camera.cpp:711 Camera in Running state trying release() requiring state between Available and Configured`

# Enviroment
Raspberry Pi 4B 4GB
IMX477 official HQ Camera
Raspbian Bookworm 64-Bit Lite build
Python 3.11.2
Flask 3.1.0

kbingham, 2025-04-04:
> [0:02:18.975520820] [1024] ERROR Camera camera.cpp:711 Camera in Running state trying release() requiring state between Available and Configured

That's curious - the camera is being destroyed or released before being stopped. I expect in this scenario we could introduce an implicit 'stop()' on the camera in the state machine in libcamera. I'd be curious to see how this situation was generated, but this is an interesting test case to add to libcamera.

signag, 2025-04-04:
> configured camera via config section:

Hi,
thanks for posting this,

Just to be clear: for which of the configurations('Live View' or 'Photo') did you set the configuration.
I assume its for the 'Live View', because for 'Photo' the YUV420 Stream Format would lead to an error ```Stream format YUV420 not supported for PIL images``` when trying to take a photo.

I tried this configuration with the same setup as you.
When trying to take a photo, the server did not return control but I did not get the error message.

Nevertheless, there is a conflicting situation:
- for the Live View, for which you have set the 'main' stream, the format is YUV420 (this is, in principle, allowed and works for the stream, but not for PIL images)
- for the 'Photo' configuration, I assume that you did not change the default, so this will also use the 'main' stream but with Stream Format "RGB888".

When configuring the live view, the status line should have shown the     
```WARNING: If you do not set Stream to 'lores', the Live Stream cannot be shown parallel to other activities!```

Since the 'main' stream cannot be used simultanously with different configurations, raspiCamSrv should
1. stop the live stream thread
2. set the configuration for photo taking
3. take a photo
4. set the configuration back to that for the live stream
5. start the live stream thread

I think, the camera might also be closed and reopened in this process.

Assumingly, something went wrong there.    
I need to analyze in detail.

In the meantime, you could try with identical configurations for Live View and Photo:
- Colour Space: Rec709
- Sensor Mode: 0
- Stream: main
- Stream Format: RGB888

Then you will get a high quality live stream and photo taking will work as well because raspiCamSrv will recognise that a photo can be taken while the live stream remains active..

wb1016, 2025-04-04:
ok. i found out that using stream 'main' was the problem. setting stream mode as 'lores' and format as 'YUV420' gave me same behavior. but doing so gives me major quality issue as mentioned in #37 .
in summary, only 'lores' stream can be used in parallel processing but i can only use 'main' stream due to quality issue.
so i suggest making streaming feed to stop processing while taking snapshot so parallel processing is not needed.

signag, 2025-04-04:
> ok. i found out that using stream 'main' was the problem.

Not really.   
The problem was that you have chosen a different Colour Space and a different Stream Format for the 'main' stream in the **Live View** and **Photo** configurations.

But of course, this sould work too. I'll fix that.