signag/raspi-cam-srv #55: ERROR in camCfg ... config/serverConfig.json: 'NoneType' object is not subscriptable

lidgaca, 2025-04-04:
I've just built this today with latest bookworm / Pi 4B 4GB / HQ camera. Thanks so very much for this project - it is just what I'm looking for.

One slight problem, After a clean build, I connected to the server via firefox and went to the config page. I changed the live feed stream to 'main' and RGB888 (as I have a pi 4 and without that the live feed data stream is rather poor quality). 

Then I went into the auto-exposure controls, setting auto-exposure 'enabled', metering mode to be 'Centre weighted' and the constraint mode to be 'Shadows'. I set the Image AWB to be 'enabled' and AWB mode to 'Daylight'

Then I saved the configuration, and set 'Start server with stored configuration'.

Now when I start the server I see an error message ...

[2025-04-04 17:26:24,549] ERROR in camCfg: Error loading from /home/pi/server/raspi-cam-srv/raspiCamSrv/static/config/serverConfig.json: 'NoneType' object is not subscriptable
 * Serving Flask app 'raspiCamSrv'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:5000
 * Running on http://10.10.10.92:5000
Press CTRL+C to quit

Things seem to work ... but is this something to be concerned about ?

-- Chris

signag, 2025-04-04:
Hi 
thanks for posting this.

I could reproduce the error and will fix it.

Since I am working on a new version where the affected file is also modified, I cannot commit the fix until I'll publish the new version.
Probably, this will be tomorrow.

Sorry

signag, 2025-04-05:
Hi,   
It may take a little bit longer with the fix.

The error occurs because during loading of stored configuration data, the system tries to instantiate a [device](https://github.com/signag/raspi-cam-srv/blob/main/docs/SettingsDevices.md) while there is none defined.

You can avoid the error by creating an arbitrary [device](https://github.com/signag/raspi-cam-srv/blob/main/docs/SettingsDevices.md) in the *Settings/Devices* dialog.    
You do not need to configure it.   
However, you need to store the configuration afterwards.