signag/raspi-cam-srv #58: user & password to clone ---raspi-cam-svr

OldPlumber, 2025-05-16:
Hello,
I'm working through "RaspiCamSrv Installation" to step 5

git clone https://github.com/signag/raspi-cam-srv

I am asked for Username and Password for https://github.com

What should I enter here?

Thanks, in advance,
Stephen

signag, 2025-05-17:
Hi,
the repository is public. So, cloning should not require registration. I do Not know why you are asked for credentials. If you register at GitHub, you could use these credentials.

OldPlumber, 2025-05-19:
Thanks for your response.
I tried again today and the cloning worked ok without using my credentials.
The only difference may be that the last time I was NOT going headless to the RPi Zero 2W.
This time I accessed via SSH,
Would this have made the difference?
Stephen.